hzeller/rpi-rgb-led-matrix #1236: youtube-dl error

rijotech, 2021-01-19:
Hi, I'm trying to get youtube-dl to work but all I get is the error below.

```
sudo youtube-dl -o- -f"[height<480]" "https://www.youtube.com/watch?v=Hkc3hXyvbfQ" | sudo video-viewer --led-rows=64 --led-cols=128 --led-chain=4 --led-parallel=2 --led-pixel-mapper=V-mapper --led-slowdown-gpio=3 --led-gpio-mapping=regular --led-rgb-sequence=RBG --led-pwm-lsb-nanoseconds=85 /dev/stdin/
Issue opening file: : Not a directory
[youtube] Hkc3hXyvbfQ: Downloading webpage
[download] Destination: -
[download]   0.0% of 95.54MiB at  1.22MiB/s ETA 01:18

ERROR: unable to write data: [Errno 32] Broken pipe
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
BrokenPipeError: [Errno 32] Broken pipe
```
I'm wondering, how can I fix this?

hzeller, 2021-01-19:
You have a slash at the end of your `/dev/stdin/` but you probably want to have it without that.

Not sure if this solves the problem, but at least this is preventing more currently.

rijotech, 2021-01-19:
I tried that, but I still get an error message.
```
sudo ~/bin/youtube-dl -o- -f"[height<480]" "https://www.youtube.com/watch?v=Hkc3hXyvbfQ" | sudo ./video-viewer --led-rows=64 --led-cols=128 --led-chain=4 --led-parallel=2 --led-pixel-mapper=V-mapper --led-slowdown-gpio=3 --led-gpio-mapping=regular --led-rgb-sequence=RBG --led-pwm-lsb-nanoseconds=85 /dev/stdin/
Issue opening file: : Not a directory
[youtube] Hkc3hXyvbfQ: Downloading webpage
[download] Destination: -
[download]   0.0% of 95.54MiB at 914.85KiB/s ETA 01:47

ERROR: unable to write data: [Errno 32] Broken pipe
```

hzeller, 2021-01-19:
Does it work when you write it to a file first and then load with the video-viewer ? I think I never tested to read a video from stdin.

rijotech, 2021-01-19:
Basically, I assumed it was going to work because of this past issue https://github.com/hzeller/rpi-rgb-led-matrix/issues/1006
I tried different methods but nothing works.

hzeller, 2021-01-20:
Did you verify first that youtube-dl does actually work and outputs a video file ? This often breaks.

Make that work first. Then do the piping.

rijotech, 2021-01-20:
When I execute the following command `~/bin/youtube-dl --no-continue -o- -f"[height<480]" "https://www.youtube.com/watch?v=Hkc3hXyvbfQ"` , the terminal does some weird stuff, I'm assuming is streaming the file. 
https://youtu.be/Tt2JxdnpdIQ

hzeller, 2021-01-20:
try to output to a file instead. The `-o` option is the one that gives the output, currently you choose `-` which means 'stdout'.

So if you use `-ovideofile` then you can later try to play that file with the video viewer.

rijotech, 2021-01-26:
I give up I'll try something different some other time with a batch script. Thanks for your help Mr, HZeller

closed 2021-01-26 by rijotech