hzeller/rpi-rgb-led-matrix #1240: Right-to-left scrolling and blinking text arabic language

manuel1507, 2021-01-26:
Hi, I'm using the library to display text on an information sign. I would need to show scrolling text in arabic language. The example scrolls from right to left. Can you explain me the changes to the code in order to scroll from left to right and also to make a blinking text, please? Really appreciated! Best Regards

hzeller, 2021-01-26:
I've added scrolling in both directions now for the [`text-scroller`](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#text-scroller) utility. If you provide a negative speed, it scrolls in the other direction.

Blinking is now provided with the `-b` option. You can choose on and off time separately; for instance, with a value of `-b10,3` for each 13 pixels it scrolls, it will be on for 10 and off for 3.

manuel1507, 2021-01-27:
Wow!! Thank you very much! That was quick! BUT, there's always a "but" :-) would be very nice if blinking would work also with speed 0! Appreciated.

anandrajgupta, 2021-08-10:
Hi @manuel1507,
I am working on a project requiring multilingual support.
Can you please tell me how are you able to show "arabic" text.
Thank you...