hzeller/rpi-rgb-led-matrix #1242: 64x64 HUB75E Issue

hoachin, 2021-01-31:
Hi,

I'm using an RPi 3B+ connected to a 64x64 panel via an active adapter board. I'm using a 5V 15A supply to this.

On the adapter board I have a jumper to enable P8 and another which ties P4 to GND. 

I was trying to run the demo program with the following:
`sudo examples-api-user/demo -D0 --led-cols=64 --led-rows=64`

However this was just displaying a blank screen.

I then updated this to include a panel type (which I guessed as being FM6126A), i.e:
`sudo examples-api-user/demo -D0 --led-cols=64 --led-rows=64 --led-panel-type=FM6126A`

Now I can see the demo running, however as you can see from the attached image the colours don't look quite right and half of the display has all the blue leds on.

I've played around with all of the various other options available but can't seem to get this to work.

I can confirm the panel isn't faulty as I've also tried it with a Teensy3.6 and the SmartMatrix library - this did require a reset sequence to be run as they discussed [here](https://github.com/pixelmatix/SmartMatrix/issues/78). 

I tried to port [a copy of the original panel reset sketch ](https://bobdavis321.blogspot.com/2019/03/teensy-31-and-smartmatrix-sd-running.html) to something that could run on the RPi but either I didn't do it right or I was being too optimistic that it would be that simple.

I'm not sure if there is anything I may have missed or next steps to try

[image]

hoachin, 2021-01-31:
I've just realised I may have got the origin of the reset script backwards as it appears to have originally been discussed as part of this library [on this issue](https://github.com/hzeller/rpi-rgb-led-matrix/issues/746), but running the reset script attached to that I get a single row of red leds and a single row of (bright) blue leds.

Running the example after running this reset script still gives me the same issue as show above

hrocha1, 2021-02-08:
@hoachin Do you have the same issue without the active adapter board or on the other connectors/chains of the adapter board?

I had similar issue once (second half of my display was green). The problem was that my adapter board wasn't working properly (first time soldering such a tiny chip) and one of the green channels on that chain was always high.

hoachin, 2021-02-21:
@hrocha1 

I've (finally) got around to testing this again. It looks like you're right though, there is something wrong with my adapter board that is causing this issue, though I haven't gotten to the bottom of what exactly that is I've got some (pre made) adapter boards coming soon so I probably never will.

closed 2021-02-21 by hoachin