hzeller/rpi-rgb-led-matrix #1243: RGB 32x32 LED's aren't displaying anything with Bonnet

milajo2, 2021-02-01:
I'm really sorry if this comes off very simple. I've recently installed and followed the tutorial for this using the adafruit installation, however, everytime I run the program nothing is displayed on the LED's. I've attached images on this post. I am using a raspberry pi 3b, 5V 4A DC power supply, 32x32 RGB LEDs and an RGB Bonnet. I've also attached an image of the first stages of the installation as the values are different from what is stated in the tutorial. 
[image]
[image]
[image]
[image]

milajo2, 2021-02-01:
Furthermore, I'm a beginner with using raspberry pi any help would be appreciated.

hzeller, 2021-02-01:
If you use the Adafruit fork of this repository, please talk to them, I don't know what they're doing in their install script.

(Also, I suggest to not use a UI on the Raspberry Pi, but rather connect to it via ssh.)

milajo2, 2021-02-01:
> Adafruit fork

Hi, 

I use this tutorial I believe it's the same as the one stated here. 
https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi/driving-matrices

hzeller, 2021-02-01:
In the screenshot it shows that you're using the Adafruit installer script.
This project has nothing to do with Adafruit, they are just using it. So if you have trouble with an Adafruit product, you should ask them.

If you want to use this project, make sure to read https://github.com/hzeller/rpi-rgb-led-matrix

For instance in your second screenshot, it shows that you're using the hardware mapping used in this project ('regular'), while if you use an Adafruit hat, you [need to pass the command line option `--led-gpio-mapping=adafruit-hat`](https://github.com/hzeller/rpi-rgb-led-matrix#if-you-have-an-adafruit-hat-or-bonnet)

milajo2, 2021-02-04:
Hi, 
I've uninstalled the adafruit and got the GitHub one installed but everytime I try to run the demo it doesn't work.
[image]

milajo2, 2021-02-04:
It also states that example-api-use does not exist but I've checked my files it does 
[image]

milajo2, 2021-02-04:
Hi I've added more pictures for clarification 
[image]

[image]
[image]

milajo2, 2021-02-04:
Hi, 
I've reinstalled everything I've now got the code running, but everytime it loads nothing appears on LED board.

hzeller, 2021-02-04:
what options do you supply ?

milajo2, 2021-02-04:
The commands I place are cd rpi-rgb-led-matrix cd example-api-use sudo ./demo -D0 runtext.ppm 
I've tried it with regular wiring using the updated instructions and my bonnet --led-gpio-mapping=adafruit-hat

milajo2, 2021-02-04:
I can get my LEDs working using my Arduino is there a possibility that I could run my Arduino using my raspberry pi and also add a touchboard sensor on my raspberry pi and get the two collaborating with one another?

davemaster, 2021-03-15:
Greetings,

So, you have the adafruit interfacing board? What model.
TRY wiring the "regular" pinout, and try again... may be, adafruit board is broke

Best regards

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin