hzeller/rpi-rgb-led-matrix #1244: Support for P2 160x80 panel

markphelan, 2021-02-04:
I've acquired a sample P2 panel that's 160x80 pixels (320mmx160mm). 

However, I can't specify 80 rows for the panel, is there any way to override this or do I need to do something like telling it the panel is 80x40 and they are chained 2x2?

rijotech, 2021-02-24:
@markphelan don't you think it will be useful to upload some images of the led module you just mentioned?

markphelan, 2021-02-24:
@rijotech Well, it's a panel as I've described. If you feel a picture would help then here's the chipset on the back:
[image]

I don't think there's much use taking a photo of the other side's 12,800 LEDs. It's 320x160mm, 160x80 LEDs like I stated.

In all honesty I expect this FM6363 chipset is designed for more than 64 rows and is controlled differently. My P5 works perfectly. This panel gives a brief flash during boot of the pi but displays nothing with any of the configurations I've tried.

Anton-always-junior, 2024-01-05:
Hi. Is there a chance to control FM6363-based matrix with rpi-rgb-led-matrix library? Maybe exist other solution to control standalone matrix of this type with Pi?

dev-markus, 2024-03-21:
I have also the "P2 panel - 160x80 pixels" and with the instruction example i can not add the parameter: ...--led-rows=80 --led-cols=160. It show a limitation message from libary.

Exist a way to run with this libary the panel with more resolution?

closed 2025-11-27 by marcmerlin