hzeller/rpi-rgb-led-matrix #1247: make led-image-viewer: fatal error

bennydcfc, 2021-02-08:
Hey guys,

I'm a bit of a noob, I am therefore sorry if I miss any important info out, or if I've made a clear and easily avoidable mistake.

I'm trying to run the **make led-image-viewer** in terminal, but I get the following error:

[image]

I managed to **run sudo apt-get update && sudo apt-get install libgraphicsmagick++-dev libwebp-dev -y** successfully, but it appears I am falling at the final hurdle. 

Any ideas on what to do would be most welcome :)

Cheers guys, stay safe.

hzeller, 2021-02-08:
It doesn't find `GraphicsMagick++-config`, which should've be provided by the installation of `libgraphicsmagick++-dev`.

However, maybe they changed something in the distribution; could you try to find somewhere if that changed ?

bennydcfc, 2021-02-08:
I'll try, thanks for the tip! Where would I put the file? 👍

bennydcfc, 2021-02-08:
No luck in finding GraphicsMagick++-config, is there a way of installing an older version of libgraphicsmagick++-dev?