hzeller/rpi-rgb-led-matrix #1248: Same static display whatever I send

craigerrington, 2021-02-11:
Hey folks, I'm using one of the adafruit 32x32 panels, with an RPi-Zero. There are no hats/bonnets involved.

I'm fairly sure (checked a bunch of times now) that I'm wired up as per the wiring diagram (but I'm willing to look here again!).

I've compiled the utilities, and tried them all out along with the demo. The demo and the utilities all produce the same output on the panel.

I initially thought I wasn't providing enough power, but I'm now definitely sending 4A at 5V, so I think that's good.

Any ideas appreciated.

[image]

davemaster, 2021-02-11:
Greetings,

Looks like a wiring problem.
1. Choose an interface: regular, adafruit, etc
2. Check hzeller rpi-rgb-led-matrix wiring/pinout according the interface
you choose in 1
3. Voilá

craigerrington, 2021-02-11:
Thanks.. I do suspect I'm doing something wrong. I'll go through the wiring again.

But as best I can tell so far I am following the "regular" wiring setup. I'll go through it again and check it again.

Thanks for confirming it looks like I'm messing up the wiring!

davemaster, 2021-02-11:
Ok,

Then, You must tell demo/imageview you are using "regular", tell
rpi-rgb-led-matrix you are using "regular" pinout:

demo [.... other options] --led-gpio-mapping="regular", maybe in the
compilation default, the default is not compiled with "regular" but
"adafruit" or other

davemaster, 2021-02-11:
the options in detail here
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/README.md

craigerrington, 2021-02-11:
Thanks again. The problem was my wiring.

I'd followed the guide here: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

But I wired up as per the table at the bottom which isn't the right thing to do. Now that I've followed the top diagram, everything is up and working!

Thanks for your help.

closed 2021-02-11 by craigerrington