hzeller/rpi-rgb-led-matrix #1249: Chain 64x64 instead of 128x32?

rgrech35, 2021-02-13:
Hello,
I've been playing around wiht some other packages that use rpi-rgb-led-matrix (namely the NHL and MLB led scoreboard projects).

I have two adafruit 64x32 panels that i have chained together. When I use --led-rows=62 --led-cols=32 --led-chain=2, this seems to force a 128x32 mode rather than a 64x64 mode. I don't know if I'm using the wrong configuration opions or if rgbmatrix always views chaining as rectangular instead of square. 

Any suggestions are appreciated thank you.

RockKeeper, 2021-12-08:
Hey! 
I've the same Problem. But i've using the go binding. 

I cannot get this two panels work as on big 64x64. I've tried every combination of --led-cols, --led-rows, --led-chain and --led-pixel-mapper. 

Due to inactivity on the go binding repo i thought i can try here to find some help?

bierbarrel, 2022-08-18:
I know this post is old but has anyone figured this out?

anandrajgupta, 2022-08-18:
> Hello, I've been playing around wiht some other packages that use rpi-rgb-led-matrix (namely the NHL and MLB led scoreboard projects).
>
> I have two adafruit 64x32 panels that i have chained together. [...]

64*

bierbarrel, 2022-08-18:
I ordered 2 of the active-3 hats a couple hours ago. I currently have the adafruit bonnet and I have had zero luck getting 4 32by64 panels to do 64 by 128. I have tried doing the solder mod to both the number 8 pad and the number 16 pad still no help. I have also tried about every switch configuration possible and still no luck.  Im going to solder pins to the adafruit board so that I can change back and forth from the 8 and 16 pad on the fly.

anandrajgupta, 2022-08-18:
Try --led-pixel-mapper="V-mapper"

More information in Readme

bierbarrel, 2022-10-02:
> Try --led-pixel-mapper="V-mapper"
> 
> More information in Readme

I did. It was no help. I do have the Active-3 hats now. I'll give them a try in the next few days. Thanks...

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828