hzeller/rpi-rgb-led-matrix #1250: Lag when it comes to displaying logos with "blurred edges" vs logos that are just "all pixels"?

justinodunn, 2021-02-13:
Hey all! So I'm working on a stock ticker with 6 daisy chained 64x32 LED matrix panels with the RPI 4, and how it works is that basically it takes a whole bunch of different stock prices with their different logos and generates a very wide scrolling image to display. 

Right now, I'm getting some issues with performance though, sometimes I'd get lag / stuttering with the scrolling animation, which I noticed in particular when it starts displaying logos that have "blurred edges" to them, hence some of the LED pixels end up being half lit to make up those blurred edges. However, when logos that are fully just pixelated come out, they seem to lag less. Please look at the images below to see what I'm referring to:

This is a logo that is "fully pixelated" (no LED pixels are semi lit):
[image]

This is a logo that has "blurred edges" (some LED pixels are semi lit to display those blurred edges, these logos seem to cause lag issues):
[image]

Any ideas how to fix this? Thank you.

hzeller, 2021-02-13:
The LED matrix code should not have any issue with blurred images or not.

I can only assume that your code that assembles it might need more resources to render these.

(Unrelated: I notice that the stock prices are written in an antialiased font. I'd suggest to use a bitmap font here, because you really want a crisp display of these.)

justinodunn, 2021-02-13:
Thanks for the quick response Mr. Zeller! So you don't think blurred images should affect the lag issue that much, and that it's rather something else that's causing this? Are there any flags I can use to perhaps better increase performance? I'm currently using a GPIO slowdown of 4, rows=32, col=64, chain=6, and GPIO mapping=adafruit-hat. 

Could you also please elaborate more on needing more resources to render? Sorry I didn't write the code for this, a friend made this for me.

Thank you.

hzeller, 2021-02-13:
Connecting a single chain of 6x 64 columns long panels is somewhat long, this will not give you a good refresh rate and I'd not recommend such many pixels on one chain to start with.

Make this [three chains](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/adapter/README.md) with two panels chained each and to physically arrange them in one row you can [use the VMapper (there is an example with one row)](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#v-mapper-and-vmapperz-vertical-arrangement)  - that way it presents to your application as the usual 384x32 pixel canvas.

While you use the `adafruit-hat`, you will get brightness glitches unless you do the [Adafruit-hat PWM hardware mod](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) and then run it with `adafruit-hat-pwm`.

Other than that, use the usual recommendations found in the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting), make sure to have the [isolcpus setting](https://github.com/hzeller/rpi-rgb-led-matrix#cpu-use), don't do resource-intensive stuff on the system (e.g. don't run a UI, don't have other intensive background processes running, preprocess stuff you can (e.g. if you need to scale down an image, like the logos, do it once in the beginning an then use that later),  strongly prefer C++ to Python). It is also noticeable that network traffic can cause glitches, as apparently inside the Pi this interferes with the GPIO output.

Using an antialiased font could also result in more resource usage as it it much more expensive to scale it to the proper size. Also given that it is more 'ugly', changing this to use a bitmap font is probably a good start (There is bitmap text drawing [in the graphics API of this library](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h#L109-L118)).

justinodunn, 2021-02-13:
Sounds good, thank you for the response! I will look more into this and experiment around. Will close this for now thanks!

closed 2021-02-13 by justinodunn