hzeller/rpi-rgb-led-matrix #1252: Black flickering ... HUB75E + bonnet ...

alucardxs, 2021-02-15:
Hello everyone,

I've got a HUB75E on a pi zero W.
I did the trick to avoid flickering (soldering on the bonnet) but even if the main flickerings have gone, there are some very annoying (black bars appearing every second somewhere on the screen) ... Is there a way to get a proper display ?

I'm using this command line btw :
sudo ./rpi-fb-matrix/rpi-fb-matrix --led-brightness=50 --led-slowdown-gpio=0 --led-gpio-mapping=adafruit-hat-pwm --led-show-refresh --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=300 ./rpi-fb-matrix/matrix.cfg

Thanks for your help.

hzeller, 2021-02-15:
The rpi-fb-matrix is something from Adafruit, and it looks like they are using this library, but use it as submodule that is three years old. You should try to update the rpi matrix submodule or contact Adafruit to ask them to keep it up-to-date.

In general, if you have a Pi Zero, it only has one core. You need to make sure to not use 100% CPU just for updating the matrix, otherwise the operating system will have to interrupt it to get anything else done, which is probably why you see the 'blackouts'; there is simply not enough CPU time available. It also depends on how many LED panels you connect to it of course. I'd guess you will not be able to get much more than say 128x32 pixels driven with this Pi and still get an acceptable refresh rate. What is your panel set-up ?

Try to tune the parameters (pwm bits, pwm nanoseconds, pwm-dither-bits,... ) and watch the CPU time with top. In particular on a Pi Zero, you want to make sure to not be much more than 80-90% CPU, otherwise you will get black-out lines. For instance the led-pwm-bits=7 looks like a pretty CPU intensive setting; higher settings there will allow for more 'breathing room'.

alucardxs, 2021-02-15:
Thanks for your replay,
I's a 64*64 panel (just one)
I tried with led-pwm-bits=11 and scanline modes it feels better but still blackouts.

closed 2025-11-27 by marcmerlin