hzeller/rpi-rgb-led-matrix #1256: Sound module incompatiblility

jax200, 2021-02-18:
Hi Henner

I am using your fantastic library to experiment with displaying speech-to-text, initially with some success.   I note the sound board incompatibility, so I am now using the --led-no-hardware-pulse statement with a little flicker.  The warning states that an external USB sound adapter could also be used.  Could you educate me as to what piece of hardware that might be?   I purchased one advertised on Amazon but no luck.  It was a simple USB-stereo plug device.  

[https://www.amazon.com/gp/product/B08LK4Y7CF/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1]

Jack

arahasya, 2021-03-03:
did you change the alsa configuration ?

jax200, 2021-03-04:
to the particular sound device?  yes

arahasya, 2021-03-04:
Anyway this library doesn't support sound with video yet. It is not configured

davemaster, 2021-03-04:
you can use sound from other application parallel using rpi-rgb-led-panel.

./demo --led-no-hardware-pulse ........ ............. ...........

Check your sound configuration in config.txt raspbian file

jax200, 2021-03-05:
As noted in my initial comment, that is what I am doing.  The library states this can be avoided by use of an appropriate external sound adapter.  So as I originally asked, what hardware satisfies this so as to eliminate any flicker?

arahasya, 2021-03-07:
Almost any usb sound adapter
I use this one
https://www.amazon.com/Sabrent-External-Adapter-Windows-AU-MMSA/dp/B00IRVQ0F8/ref=mp_s_a_1_1_sspa?dchild=1&keywords=Usb+sound&qid=1615101183&sr=8-1-spons&psc=1&smid=A29Y8OP2GPR7PE&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEzVUI0U1RFWkZKNDhWJmVuY3J5cHRlZElkPUEwMDMwMzA2MTZTWU9SVFZSWURJRCZlbmNyeXB0ZWRBZElkPUEwNDE1MTc5Mk5SMjBITlhaTkRWVCZ3aWRnZXROYW1lPXNwX3Bob25lX3NlYXJjaF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl

jax200, 2021-03-09:
I think I tried something similar, but it did not eliminate the flicker.

[https://www.amazon.com/gp/product/B08LK4Y7CF/ref=ppx_yo_dt_b_asin_title_o03_s01?ie=UTF8&psc=1](url)

arahasya, 2021-03-09:
what arguments are you using to run the library?

jax200, 2021-03-10:
Thanks @davemaster and @arahasya.  I tweaked config.txt and it appears to have knocked down the flicker a bit.

closed 2025-11-27 by marcmerlin