hzeller/rpi-rgb-led-matrix #1261: Minor flickering problem after 20 minutes

rijotech, 2021-02-28:
Hi Mr. Zeller. 
I'm facing a minor issue with a screen, I'm running 8 P2.5 128x64 LED modules, I have also added 8 capacitor 25v 3300uf, I noticed that when the screen is off for a few hours and I turn it on there is no static or flickering but after 10 to 20 minutes I start noticing some flickering on the left upper last two modules also depending on the color of the video the flickering gets more intense.

So far I replaced those two modules, power supply, shorted the power cables without any luck. 

Can you please provide me with some possible solutions?

davemaster, 2021-02-28:
try reduce --led-slowdown-gpio=(2-4) also  --led-pwm-lsb-nanoseconds=(200-350); in this last, there is a way to try the better values for this, using refresh rates values (make them visible)

rijotech, 2021-02-28:
@davemaster I have tried all that and --led-slowdown-gpio=(2-4) will just make it worst, I don't think this is a command-related problem but I could be wrong, I will try using capacitor 25v 4700uf or 35v 3300uf to see if that makes any difference.

rijotech, 2021-03-08:
Well, the capacitor 25v 4700uf didn't work if any I got even more static, I'll try using ground and power bars next.

davemaster, 2021-03-09:
Did You try with another RGB LED panels?

rijotech, 2021-03-11:
@davemaster Yep shorted the cables, change the power supply, grounded the raspberry pi, swap capacitors, disable crontab, disabled all running services, disabled audio on boot.txt dtparam=audio=off, applied isolcpus=3 on /boot/cmdline.txt and still I can't get rid of the flickering.

rijotech, 2021-03-11:
I just fixed it finally.

closed 2021-03-11 by rijotech

hzeller, 2021-03-11:
What was the issue ?

rijotech, 2021-03-11:
The data cable was too long I purchased different cables [image]