hzeller/rpi-rgb-led-matrix #1262: Rpi-rgb-led-matrix with Raspberry PI4 with tutorial : command not found

SebastienLBA, 2021-03-01:
Hello,
I am a real beginner with Raspberry Pi and all this "coding" thing ! I am mostly an electrical engineer and I used to program PLC with functional block. 
But, for a project, I need to drive a matrix of leds, which is impossible with the PLC I am currently using so I thought about using a Raspberry Pi.
And here I am ! I bought a Raspberry Pi4 and a matrix of leds and I try to follow this tutorial for beginner : 
https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/overview

I think I did everything well until the step "testing" : 
https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/testing

But I must be missing something because, after the final command `sudo ./led-matrix`, it writes the following message : 
`sudo ./led-matrix : command not found`
Here is the full printscreen : 
[image]

I don't know what I am missing, certainly something dumb but I 
I think it may be about the file "Makefile" because I do not find the line that I have to uncomment : 
`DEFINES+=-DRGB_CLASSIC_PINOUT`

Anyway, I am calling for your help because I don not have a clue and I try to figure thie out for 2 weeks now...

Thanks a lot,

Seb

davemaster, 2021-03-01:
FIRST, you need more knowledge about linux raspbian environment: us ls command to see files (as dir in DOS)

SECOND, the adafruit tutorial about how to testing LED panel, is very OLD, the rpi-rgb-led-matrix is continously updated; the command/program/executable led-matrix is not included anymore.

THRID, did you build the executables using the compiler? READ the README.md file FIRST....

tHEN use:
sudo ./demo

hzeller, 2021-03-01:
+1 What David said.
And the demo is in examples-api-use. The relevant section in the readme is https://github.com/hzeller/rpi-rgb-led-matrix#lets-do-it

@SebastienLBA Please also write the Adafruit people that they should update their tutorial about my library.

SebastienLBA, 2021-03-02:
Thanks a lot for your quick answers.
I will try with this new tutorial.

SebastienLBA, 2021-03-03:
Hello, It's me again,
So I followed the the readme in https://github.com/hzeller/rpi-rgb-led-matrix#lets-do-it and almost everything went well.
Everything seems to be well connected but I struggle to have a good rotating square with the demo -D0 (pictures below).
There are plenty of other leds ON and kind of ghosting.

I am using the following led-matrix : 
[https://shop.mchobby.be/fr/leds/1318-matrice-led-rgb-16x32-empattement-5mm-3232100013186.html](url)
There is not much details in the datasheet.

Here is my standard command : 
`sudo examples-api-use/demo -D0 --led-rows=16 -- led-cols=32 --led-chain=1 --led-parallel=1 --led-no-hardware-pulse`

I tried to play with every parameters such as : 

- `--led-gpio-making`
- `--led-multiplexing`
- `--ledpixel-mapper`
- `--led-pwm-bits`
- `--led-scan-mode`
- `--led-pwm-dither-bits`
- `--led-limit-refresh`

Nothing works fine, the best parameters are always the parameters by default.
Do you know what could possibly be the problem?
Thanks for your help,

Seb

Here are the pictures : 
[image]
[image]

hzeller, 2021-03-03:
Need more detail :) For instance, did you go through the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting) and made sure to follow all steps there ?

Did you [wire it directly](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) to the Pi or used [digital level converters](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter) ?
Not all panels will work with 3.3V logic level and if so it might be a hit and miss if wired directly.

If you wired it manually, it is particularly important that the wires between the Pi and the panel are short.

Did you double-check that GND is well connected between Pi and Panel ?
Where is the power coming to the Matrix coming from ? It needs to be a good power supply that can deal with short spikes. Make sure the power cable from the power supply to the Panel is direct and as short as possible. Consider [adding a low ESR capacitor](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power) close to the panel.

The most influential parameter in this scenario will probably be `--led-slowdown-gpio`. The Pi4 is fast, and panels typically can't keep upt. Start with `--led-slowdown-gpio=4` to see and reduce maybe to 3 if you panel can keep up.

Don't use `--led-no-hardware-pulse`; it is only really meant to work-around situations where it can't be used, but it will provide less stable color output.

SebastienLBA, 2021-03-04:
Thank you very much !!!! It was the only parameter I did not try! It works very well with `--led-slowdown-gpio=2` !

For all your others wuestions, Everything is OK, I have a good power supply, with short wires. And I wired it myself but with good and short wires :-)

Thanks you, now that I have the basement, I will try to have some fun with it !

jaygarcia, 2021-04-09:
@SebastienLBA please close out this issue when you have a moment :).

koplinm, 2024-01-07:
I've run into the same issue, this thread is helpful.  The program I am trying to run is calling demo.  Is there something similar to demo that does not require the manual ctrl C reset for new display content?