hzeller/rpi-rgb-led-matrix #1264: [question] option 'D' doesn't work(32x16, it show only 4 line)

JibeomPark, 2021-03-04:
[image]

./demo or any  commands are show same result.
what is problem? How can I fix it?

JibeomPark, 2021-03-04:
@hzeller  please help me

fvzdrl, 2021-03-04:
You have to try many configurations as follows:
sudo ./demo --led-rows=16 --led-slowdown-gpio=<**0->5**> --led-multiplexing=<**0->17**> -D 1 runtext.ppm

JibeomPark, 2021-03-04:
> You have to try many configurations as follows:
> sudo ./demo --led-rows=16 --led-slowdown-gpio=<**0->5**> --led-multiplexing=<**0->17**> -D 1 runtext.ppm

thank you. I'll try that.

JibeomPark, 2021-03-04:
> You have to try many configurations as follows:
> sudo ./demo --led-rows=16 --led-slowdown-gpio=<**0->5**> --led-multiplexing=<**0->17**> -D 1 runtext.ppm

same state :( and there's new error : Error in PixelMapper(it's seem like a index error)

JibeomPark, 2021-03-04:
my gpio connect

pi - led

6 GND
7 STB
11 CLK
12 OE
13 G1
15 A
16 B
18 C
19 B2
21 G2
22 D
23 R1
24 R2
26 B1

total : 14

is it need 10-E or 22-D? I only use 1 pannel(16x32)

hzeller, 2021-03-04:
LED [`--led-row-addr-type, --led-multiplexing or --led-panel-type`](https://github.com/hzeller/rpi-rgb-led-matrix/#types-of-displays) are the most likely the needed setting, in that order of likelihood.
And yes, some multiplexing options won't work for the panel size and will create an error, so that is fine.

It is unlikely that you need to connect `D` with this panel size.

If you connect it directly without digital level shifter, then this might also be a cause (not all panels can deal with the low 3.3V logic levels from the Pi, but you already have seen that in the [wiring instructions](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) doc.

JibeomPark, 2021-03-04:
> LED [`--led-row-addr-type, --led-multiplexing or --led-panel-type`](https://github.com/hzeller/rpi-rgb-led-matrix/#types-of-displays) are the most likely the needed setting, in that order of [...]

First of all, thank your reply.

I bought this panel from Korean shop but it is same for Adafruit's.  And fortunately, this panel doesn't need shifter(I find this by searching internet)

Actually, I think It is not connection problem(like a GPIO or others). But I can not understand that why it's not working. I mean, why this situation happen to me?(photo)
[image]

Do you think that because I didn't use shifter this happened? Or any reason?

JibeomPark, 2021-03-04:
> LED [`--led-row-addr-type, --led-multiplexing or --led-panel-type`](https://github.com/hzeller/rpi-rgb-led-matrix/#types-of-displays) are the most likely the needed setting, in that order of [...]

There are some panels that have a different chip-set than the default HUB75. These require some initialization sequence. The current supported types are --led-panel-type=FM6126A and --led-panel-type=FM6127.

=> My panel have HUB75 type. Do I need initialization? What is the command?

hzeller, 2021-03-04:
the next sentence after initialization says it: change `--led-panel-type`.
Did you already try all `--led-row-addr-type` settings ?

JibeomPark, 2021-03-04:
> the next sentence after initialization says it: change `--led-panel-type`.
> Did you already try all `--led-row-addr-type` settings ?

not yet. I try that your old code(05/2015 and old GPIO connect to copy Adafruit tutorial. I'll change GPIO connect like docs and try that command.  Thanks your reply.

--led-row-addr-type
--led-panel-type

JibeomPark, 2021-03-04:
> the next sentence after initialization says it: change `--led-panel-type`.
> Did you already try all `--led-row-addr-type` settings ?

I tried that --led-row-addr-type<0~4> but same result(the line's number changed 4 or 2, and some line's brightness are changed, but still show just line.

hzeller, 2021-03-04:
Don't use the Adafruit tutorial or the Adafruit branch of this library. Use this library otherwise all bets are off (e.g. they change by default the hardware mapping to the board they're selling)

JibeomPark, 2021-03-04:
> Don't use the Adafruit tutorial or the Adafruit branch of this library. Use this library otherwise all bets are off (e.g. they change by default the hardware mapping to the board they're selling)

yes. Now I only use this docs.

JibeomPark, 2021-03-04:
> Don't use the Adafruit tutorial or the Adafruit branch of this library. Use this library otherwise all bets are off (e.g. they change by default the hardware mapping to the board they're selling)

Did you see same problem like this?
[image]

davemaster, 2021-03-04:
Greetings,

To START using the rpi-rgb-led-matrix library follow these steps, BUT FIRST, read the readme.md of this repository:
1. Wiring: You MUST KNOW what interface are you going to use, in DETAIL, the pinout of your Raspberry Pi GPIO, the HUB75 RGB LED PANEL and Your interface pinouts.
2. You MUST KNOW, what RGB PANEL are You bought; then, use  --led-multiplexing according the panel You have.
3. Once you did a wiring and panel recognition, try ./demo options in an adecuated way...

Best regards

JibeomPark, 2021-03-04:
> Greetings,
>
> To START using the rpi-rgb-led-matrix library follow these steps, BUT FIRST, read the readme.md of this repository:
>
> 1. Wiring: You MUST KNOW what interface are you going to use, in [...]

Thanks to reply.
I use RPI 3B+, and I boutht Adafruit 16x32 led matrix panel(HUB75, 1:8, https://www.adafruit.com/product/420 )
This penel's scan rate is 1:8.
So I use --led-row-addr-type=2 and --led-multiplexing=1 but It's not working

JibeomPark, 2021-03-04:
I use command

sudo ./demo -D 1  runtext16.ppm --led-rows=16 --led-cols=32 --led-chain=1 --led-brightness=5 --led-parallel=1 --led-slowdown-gpio=3 --led-row-addr-type=2 --led-multiplexing=1 --led-gpio-mapping=regular-pi1

hzeller, 2021-03-04:
why do you use `regular-pi1` when you wired it up according to the [wiring.md](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) ?

JibeomPark, 2021-03-04:
> why do you use `regular-pi1` when you wired it up according to the [wiring.md](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) ?

arcording to docs, regular's chain is 3, regular-pi1's chain 1. so I do that.  and 2 way results are sam

hzeller, 2021-03-04:
Where did you read that ?
The `regular-pi1` is if you have the first generation of Raspberry Pi's and want to connect it.
For everyone else, there is regular, independent of number of chains.

JibeomPark, 2021-03-05:
> Where did you read that ?
> The `regular-pi1` is if you have the first generation of Raspberry Pi's and want to connect it.
> For everyone else, there is `regular`, independent of number of chains.

I just tried regular-pi1 because 'regular' didn't work. Already I tried that....

JibeomPark, 2021-03-05:
@hzeller 
When I supply power led panel and raspberry, pi wasn't turn on, but supply pi before connect adapter to led panel, then pi turn on.
Is it affect this situation?(4 line only)

beta-tester, 2021-03-05:
@JibeomPark 
at adafruit there is written, that for a 32x16 panel pin D does not exist and must be connect to GND...
see https://learn.adafruit.com/32x16-32x32-rgb-led-matrix/new-wiring
i don't know if it has to be GND directly or if it is good when the GPIO pin of the RPi pull this pin to logic-low level.
and i don't know it possible in the rpi-led-rgb-matrix software to do that.

hzeller, 2021-03-05:
Connecting GND to at least one pin that says GND on the Matrix board to the GND of the Pi is important.
Typically, if that is the case, then leaving the empty spot on `D` disconnected is probably fine.

closed 2025-11-27 by marcmerlin