hzeller/rpi-rgb-led-matrix #1270: Power lose issue

dannielum, 2021-03-07:
Hi, I keep having the power lose issue with my raspberry pi with the Adafruit RGB Matrix HAT. I was able to get it to work by using only one 5v power adaptor connected to the HAT. However, I notice that whenever I turn on the fan or basically any device connected to a power outlet within the same room, the LED Matrix will just lose power. I was still able to ping the raspberry pi, so I am guessing it was just the LED Matrix was losing the power. I have to then restart the raspberry pi to get the LED Matrix to work again. I tried connecting both power adaptor to both the raspberry pi and the Matrix HAT thinking it might solve the power outage issue, but it didn't help and same thing happened. Does anyone else have the similar issue? Thanks!

hzeller, 2021-03-07:
Make sure that you power supply can deal with the load (and read the [word about power](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power) paragraph).

But the problems you describe just point to a not sufficient power supply or bad house wiring :)

dannielum, 2021-03-07:
Thanks for the speedy response! It could be the bad soldering, it was my first soldering from few years ago and I am trying to expand the work on this project and trying to solve this existing issue I had been ignoring since the beginning. Just wondering, do we need to connect both power adaptor or just a single power adaptor to the HAT is sufficient? I only have 1 16x8 LED matrix board.

hzeller, 2021-03-07:
If you have multiple power adapters (say, one for the Pi and one for the Matrix), you only connect the GNDs together, but have separate +5V (in fact, you _must_ separate the 5V signals)

dannielum, 2021-03-08:
Thanks for your advice, I need to review my configuration and soldering. I was pinging the wrong ip address before, so the raspberry pi actually just lose the power whenever a device is turned on from another power outlet.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin