hzeller/rpi-rgb-led-matrix #1271: How to make text static?

davidjmorin, 2021-03-10:
I am using the below code and wanted to know how I would make this static and not scroll on the screen.  It should update every 10 seconds though. 
```

from samplebase import SampleBase
from rgbmatrix import graphics
import time
import requests
from bs4 import BeautifulSoup as bs

class RunText(SampleBase):

    def run(self):
      while True:
        res = requests.get('https://www.marketwatch.com/investing/stock/gme')
        soup = bs(res.content, 'lxml')
        price = soup.select_one('.intraday__price .value').text

        res1 = requests.get('https://www.marketwatch.com/investing/stock/etsy')
        soup1 = bs(res1.content, 'lxml')
        price1 = soup1.select_one('.intraday__price .value').text

        res2 = requests.get('https://www.marketwatch.com/investing/stock/aapl')
        soup2 = bs(res2.content, 'lxml')
        price2 = soup2.select_one('.intraday__price .value').text

        offscreen_canvas = self.matrix.CreateFrameCanvas()
        font = graphics.Font()
        font.LoadFont("../../../fonts/6x10.bdf")
        textColor = graphics.Color(255, 255, 0)
        stock2 = graphics.Color(139, 0, 0)
        stock3 = graphics.Color(64, 0, 255)
        pos = offscreen_canvas.width
        my_text =  " GME: " + price
        my_text1 = "ETSY: " + price1
        my_text2 = "AAPL: " + price2

        while True:
            offscreen_canvas.Clear()
            len = graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, my_text)
            len2 = graphics.DrawText(offscreen_canvas, font, pos, 18, stock2, my_text1)
            len3 = graphics.DrawText(offscreen_canvas, font, pos, 26, stock3, my_text2)
            pos -= 1
            if (pos + len3 < 0):
                pos = offscreen_canvas.width
                break
            time.sleep(0.08)
            offscreen_canvas = self.matrix.SwapOnVSync(offscreen_canvas)

#Main function
if __name__ == "__main__":
    run_text = RunText()
    time.sleep(0.08)
    if (not run_text.process()):
        run_text.print_help()

```

hzeller, 2021-03-10:
Just don't update the position.

davidjmorin, 2021-03-10:
> Just don't update the position.

So remove this?
```

            pos -= 1
            if (pos + len3 < 0):
                pos = offscreen_canvas.width
                break
```

davidjmorin, 2021-03-16:
What if I wanted to scroll just one line?  How would I write that up?  
So line 1 scroll
Line 2 static.

hzeller, 2021-03-16:
You have a canvas where you can do whatever you want with it, displaying things and animating things.

This library assumes that you can read and understand the provided APIs. For basic programming advice please ask your local peers who do programming, it is probably much faster than trying to do that via the issue tracker.

closed 2021-03-16 by davidjmorin