hzeller/rpi-rgb-led-matrix #1274: Can´t run a 64x64 mattrix with adafruit bonnet

Investegate, 2021-03-13:
Hello, I'm new to LED matrices.
I have now spent over 10 hours getting my 64x64 LED matrix to work. Unfortunately so far without the slightest success.

Therefore I ask for your support.

I run the matrix with a RPI 2b v1.1 and an Adafruit RGB Matrix Bonnet. The panel is from Aliexpress (https://a.aliexpress.com/_mLDLYhN)
Here the manufacturer sent me a PDF of the built-in ICs:
https://www.docdroid.net/Oy5GKYW/fm6124-pdf
Thats the pinout of the hub75:
https://i.ibb.co/gTwhjGp/Hfbbe8c4bf5a04eb29a9f0eae83d7401c-Q.jpg

I measured E and realized that I have to connect it to 8. The manufacturer was able to confirm this to me with the hub75 pinout picture.

I've already tried all of these:
- Quality mode with the 4-18 connection.
- convidence mode
- the most flags in different combinations
- a second adafruit bonnet to rule out a defect.

Now I don't know what to do next.
I have now ordered another panel drive board for the RPI.

Maybe someone else notices something that I could still try.
I would be very grateful!

hzeller, 2021-03-13:
What is the command line you are using ? Most important of course is to use `--led-gpio-mapping=adafruit-hat` if you are using the Adafruit Bonnet/Hat. Or `adafruit-hat-pwm` if you did the 4-18 mod.

There are some types of displays that need to have a particular chip initialization. So if your chip is a FM6124 (the PDF you linked), then maybe `--led-panel-type=FM6126A` might be the closest. But if that is not working, then you have to go through the datasheet and have to implement the initialization of FM6124 in the matrix framebuffer.cc and wire it up. Then send a pull request.

The locations are:

  * FM6126 https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer.cc#L471
  * FM6127 https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer.cc#L511
  * FM6124 `<implementation you might need to add>`

Investegate, 2021-03-13:
> What is the command line you are using ? Most important of course is to use `--led-gpio-mapping=adafruit-hat` if you are using the Adafruit Bonnet/Hat. Or `adafruit-hat-pwm` if you did the 4-18 [...]

Thank you for your reply!
I already know that. I've browsed a few forums already.

With `--led-panel-type=FM6126A` I have already tried many variants. but without success.

For an FM6124 implementation I unfortunately do not have the necessary knowledge to do something like this.

Which panels have been tested and recommended? I need 3 pieces of P2 64x64 panels.

Best Regards!

Investegate, 2021-03-18:
Can someone please recommend a suitable panel that works well without further ado? (P2 64x64)

britxpatusa, 2021-03-29:
> Can someone please recommend a suitable panel that works well without further ado? (P2 64x64)

i got this working..while not the cheapest quick and easy: im completely new to all of this...
https://www.amazon.com/gp/product/B079JSKF21/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1

Investegate, 2021-03-31:
Hey :) I sold my Panels with the FM6124 ICs and bought some with FM6126A ICs.
I tried it, but still nothing works. Then i used an other Raspberry Pi and it works :D Maybe the FM6124 IC´s will work also, but now its to late to test it xD

closed 2021-03-31 by Investegate

Sploitastik, 2022-03-22:
Had same issue. Some panels have the E line on pin 8 instead of 4. Just undo the solder on connecting pin 8 or 16 and connect GPIO 26 to pin 8.
[Look at this issue 1279](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1279)