hzeller/rpi-rgb-led-matrix #1275: 14 PWM Bit display displaying wrong colors.

jkip53, 2021-03-14:
Hello, I'm new to this kind of stuff so sorry if I don't know what I am talking about.

I have never tried something this difficult before and I may be too in over my head. I needed a project I could work on over my break from school and matrices seemed pretty well documented. I attempted to educate myself but I missed one pretty important detail. The panels I bought for this project are from [aliexpress](https://www.aliexpress.com/item/32757647402.html?spm=a2g0o.productlist.0.0.3528235bB0aT2k&algo_pvid=33fe07e7-1b25-415e-a058-f110a59cf543&algo_expid=33fe07e7-1b25-415e-a058-f110a59cf543-1&btsid=0b0a556716128220454568942e5ba2&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_), I missed the part about the PWM Bits only being supported up to 11. My panels are 14 bit (I think) so that is why I am guessing the colors are the complete opposite of what they should be. It seems to affect the blues and greens more than the reds. I tried putting a picture of [steve](https://photos.app.goo.gl/NxZfdwA6bvy1ntGJA) on the matrices but it turned out like [this](https://photos.app.goo.gl/SbUsWUXKoeR19jBy7). I also tried [Nyan cat](https://photos.app.goo.gl/rYrKKhafU8F6ECKG8) but it also turned out with the wrong [colors](https://photos.app.goo.gl/E6FfFNs16sSfbHWKA). I looked around a bit before I submitted this and I came across https://github.com/mankeli/hlsudp-rpi which is supposed to allow up to 16 bits according to [this](https://github.com/hzeller/rpi-rgb-led-matrix/issues/753#issuecomment-455855340), but when I tried to run a gif with it it still said that it only supported up to 11 which seemed strange, since it was supposed to get rid of that issue. I am using a Pi 4 4GB with an Adafruit HAT and a 15 Amp brick.
If I could get some help that would be great, Thank you.

davemaster, 2021-03-14:
Greetings,

With other values of PWM bit right colors are displayed? I remember there are another options different than RGB color components

Best regards

arahasya, 2021-03-14:
Hello,

Don't get confused with the shift registers of 16 bits (16 output channels) used in the module with pwm bits.

First of all, try to run the demos like pulsing colour and show the output without modifying the pwm bits argument.

jkip53, 2021-03-14:
Thank you for replying so fast,
I have been trying out different flags all morning. if I use demo 4 and don't change or add any flags for the PWM bits all the colors look great and nothing seems out of place. I have also tried using the --led-panel-type flag which made the panels stop working so I left that flag on default, I also experimented with all 17 led multiplexing and they just mixed up the image so I also left them on default. I tried the PWM bit flag 1-11 and they didn't seem to make a difference at all except for 1 which changed all the colors to dark. I am not sure what --led-pwm-dither-bits seems to do other than improve refresh rate. 
Thank you for the help.

davemaster, 2021-03-15:
Try to share with us a picture of the chips side of your panel...

jkip53, 2021-03-16:
I found 3 different chips on my panel,
[Photos.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/6145366/Photos.zip)
I have no idea what they mean.

davemaster, 2021-03-16:
> 
> 
> I found 3 different chips on my panel,
> [Photos.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/6145366/Photos.zip)
> I have no idea what they mean.

We need a picture like this, so your rgb panel can be identified:

[image]

hzeller, 2021-03-16:
Don't get hang up on the 14bit - that is just a marketing term used by the manufacturers: it is a theoretical maximum with a very low refresh rate. You can drive them with 1 bit or 11 bit - the difference is the number of different colors that can be displayed.

Your problem looks like the panel has RGB wired up differently. You can work around it by setting `--led-rgb-sequence` to map the right colors. See [README](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/README.md) for details.

jkip53, 2021-03-17:
I just tried the --led-rgb-sequence and the sequence RBG worked!
Thank you for the help I wouldn't have figured this out myself.

closed 2021-03-17 by jkip53