hzeller/rpi-rgb-led-matrix #1276: the pulsating flickering Python

6010110254, 2021-03-14:
Thanks for the library
I have a problem wanting to improve the visualization due to the pulsating flickering I use Python language. I have followed all instructions.

The board
-Raspberry Pi4
-adafruit-hat-pwm
https://drive.google.com/file/d/1yDEyO0hQAnSkTlravWUzXhDfo9bYg7Xr/view?usp=sharing

6010110254, 2021-03-14:
When showing for a while Flashing symptoms Pulsating What is the cause?
Video Example
https://drive.google.com/file/d/1SWdmroonmdDxBPE9K7aYIsj--liqNkJG/view?usp=sharing

6010110254, 2021-03-14:
@hzeller  Help me pls
thank u

macaquedev, 2021-03-30:
same problem here. But even worse in my case and it is also happening when running his demos.

macaquedev, 2021-05-03:
Fixed! You need to disable 1 Wire Protocol through raspi-config.

marcmerlin, 2025-11-30:
thank you, closing

closed 2025-11-30 by marcmerlin