hzeller/rpi-rgb-led-matrix #1277: Alternating two rows on/two rows off with matrix.SetPixel()

dripp33, 2021-03-17:
Maybe someone has also had this problem and can help? I recently purchased an Adafruit 64x64 matrix and 'bonnet' and did the E/8 and 4/18 soddering, as described. I'm running both C++ and Python examples with "--led-cols=64 --led-rows=64 --led-gpio-mapping=adafruit-hat-pwm --led-multiplexing=0 --led-slowdown-gpio=1" params and have tried all 17 --led_multiplexing options. My problem is that all examples work fine, with a leds displayed correctly, except those examples (and my own code) that use the Matrix.SetPixel() function. When this function used -- like in the game-of-life (c++) or gray-block.py (python) examples -- I get alternating bars of two correct rows of leds followed by two dark rows (y-axis); all x-axis leds are 'lit'. I've checked A.B,C,D,E, etc voltages and they are fine. Have also tried all combinations of parameters and have run out of ideas. Am almost at the point of giving up on the 64x64 display and trying a smaller 4-address matrix (although don't want to...). Hoping someone has had a similar problem and would have a suggestion...
Thanks much for any help! --dave

hzeller, 2021-03-17:
All examples use setPixel(). Did you find the difference of the ones that work and the ones that don't ?

dripp33, 2021-03-17:
Thank you for the very quick reply. After checking all the c++ and python examples you are of course correct... every example has the alternating 2 row on/off issue. I had thought that a couple of the python examples that didn't call SetPixel directly (instead call SetImage and DrawText) were working ok, but unfortunately after looking at them again not so, they also have the issue. Ugh... I was hoping that if some of the examples worked ok and some didn't it would help with diagnosing the problem. Anything you suggest I try?
Thanks again, --dave

maksun, 2021-04-09:
Hello @dripp33,

Could you please tell me what is the chipset your panel are using?

I currently face the same issue with P4 panels that use the ICN2037BP chipset (see attached picture).

[image]

In my case I'm using the electrodragon controller, I also tested many parameter combinations but never managed to get a better situation that the one on the picture ;)

Dear @hzeller, if you need me to do some new code testing that can fix this issue do not hesitate ;)

Thanks.

monoapp3, 2022-07-17:
It looks this topic is open for long time. I am using a 64x64 P3 panel. 

I can run some demos on an esp32 board by using following library:
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA#1-library-installation

With following modifications:

change number of rows and columns to 64
number of panels set to 2
connect wrong labeled pin to E control pin in esp32 board
assign number to E pin on board configuration

I tried to follow same steps with the rpi library, but without success.

Require to use a rpi due to performance topics, therefore need to use this library.
I would be really be thankful to get any hint or required configuration step.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828