hzeller/rpi-rgb-led-matrix #1278: Two colored clocks

mirgorodalexander, 2021-03-19:
Huys hi! Please help me with that thing, I wanna display clocks on 64x32 panel where hours will be one color and minutes will be second color, how I can do this?

At now I run two commands like:
`sudo ./clock --led-no-hardware-pulse --led-gpio-mapping=adafruit-hat --led-cols=64 --led-rows=32 --led-daemon --led-brightness 100 -f ./font.bdf -x 6 -y 3 -S 2 -C 0,0,0 -O 255,0,0
`
`sudo ./clock --led-no-hardware-pulse --led-gpio-mapping=adafruit-hat --led-cols=64 --led-rows=32 --led-daemon --led-brightness 100 -f ./font.bdf -x 36 -y 3 -S 2 -C 0,0,0 -O 0,0,255
`
After that I got different clock colors but hours have less brightness than minutes, why?

hzeller, 2021-03-19:
are you trying to run two different programs accessing the same panel ? There can only ever be one process that can access the gpio.

So if you are running two processes you will get them 'fighting' for the resource at best, but often just garbled output.

Just change the code to use two different colors.

mirgorodalexander, 2021-03-20:
hzeller, thankyou!

closed 2021-03-20 by mirgorodalexander