hzeller/rpi-rgb-led-matrix #1279: Yet another black bar issue 64x64 panel from Adafruit (http://adafru.it/4732)

s3rdl, 2021-03-22:
As described in the following issue: #961 #926 #895 #1189 I tried all mentioned hardware modifications on the Bonnet and the PiHAT. Even those shown on Adafruit website. I do have both boards twice, so I can depict different stages of the hardware mods.
I tried every combination of wiring and command prompt, but have no success.
This is, how I start the prog at the moment:
 `sudo ./demo -D0 --led-cols=256 --led-rows=64 --led-gpio-mapping=adafruit-hat-pwm --led-multiplexing=1 --led-limit-refresh=400 --led-show-refresh --led-pwm-dither-bits=1 --led-slowdown-gpio=4`

Used hardware: 
Pi 4 - 4GB - running raspbian buster lite (as lite as possible)
Adafruit RGB Matrix HAT + RTC for RPi (http://adafru.it/2345)
Adafruit RGB Matrix Bonnet (http://adafru.it/3211)
4x Adafruit 64x64 RGB LED Matrix - 3mm Pitch (http://adafru.it/4732) chained together

The following pictures shows the mods on the boards and the result on the LED-Matrix.

Bonnet front/back:
[image]
[image]

PiHAT front/back:
[image]
[image]

And the Matrix:
[image]

I can run static and scrolling text without any hassle, but only in the first and the third section of the panel where no black bars are.
Using larger fonts, or setting margin for text, affects in the described black bars.

Using one panel with a D1 mini ESP8266 showing a picture, all LEDs running.

Sorry to open yet another topic regarding this issue, but I have no clue anymore.
I'd appreciate any help, or hint to solve this.

Thanks in advanced

s3rdl, 2021-03-24:
OK. I found it.
In my case the E-line is on PIN8 instead of PIN4, so I soldered back PIN4 and cut the line to PIN8 and routed it to GPIO24.
Now everything works.
Cheers.

closed 2021-03-24 by s3rdl

Sploitastik, 2022-03-22:
I can confirm this works for DCHY-P2-644-1515-VP panels!