hzeller/rpi-rgb-led-matrix #1280: [question] Reading values from canvas

datagod, 2021-03-23:
Is it possible to read the values of a canvas, after it has been written to?

If I set the pixels on a canvas but want to manipulate them prior to displaying, is it possible to query the canvas at a particular co-ordinate and get the values?

example:

#create a canvas
Buffer1 = TheMatrix.CreateFrameCanvas()
Buffer1.setpixel(5,5,0,0,0)

#inside some function
#get the current rgb and add 1
r,g,b = Buffer1.GetPixel(5,5)  
r = r + 1
g = g + 1
b = b + 1

Buffer1.SetPixel(5,5,r,g,b)

I know there is no "getpixel" yet, but is there a way in python to get the information from the canvas object?

hzeller, 2021-03-23:
Unless you implement a layer on top (some canvas that just remembers the pixels) there is not.
Reason: whenever you call SetPixel(), it is internally already converted to the particular memory layout needed to send to the matrix, so it would be hard to unravel that again. 

So best would be to just build a 'shadow buffer' in which you remember the values set with SetPixel().

datagod, 2021-03-23:
Thank you for the quick response, was not expecting it.  I feel good now because my own solution was to do exactly that.  I was just going back over my code looking for performance gains and I thought I would ask.  [Here is the zooming function I built using the 'shadow buffer' and some loops.  ](https://youtu.be/Z9uW0MQYcrE?t=258)

datagod, 2021-03-23:
Good work around provided.

closed 2021-03-23 by datagod