hzeller/rpi-rgb-led-matrix #1282: 64x128 panel overlapping LEDs

I3lackshirts, 2021-03-26:
[image]

see the attached picture. Basically what is happening is the first 16 rows are displaying both the first 16 rows and rows 17-32 on top of each other. Same thing is happening for the bottom half. I tried inputting a different led-panel-type and just was getting errors. Anyone run into this before and know what to try?

davemaster, 2021-03-26:
Greetings,

This a chain or parallel or panel (a picture of its back will be usefull) issue.

What is the command are You using with this output? Can You share it?

Best regards

hzeller, 2021-03-26:
So this looks like a 64 pixel high panel. Did you invoke it with `--led-rows=64`, and, if you happen to use one of the Adafruit HATs, did the [necessary hardware modification](https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet) ?

I3lackshirts, 2021-03-26:
So I am using the Electro dragon RGB panel v2 https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-for-raspberry-pi-v2/ so I am thinking it will need a similar hardware modification? There is not much documentation about this board so not sure where to look to see where I can make a similar modification

davemaster, 2021-03-26:
> 
> 
> So I am using the Electro dragon RGB panel v2 https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-for-raspberry-pi-v2/ so I am thinking it will need a similar hardware modification? There is not much documentation about this board so not sure where to look to see where I can make a similar modification
> 
> And it is not a chain. Its one panel 64x128. I did use the commands to specify rows=64 cols=128 and no hardware pulse.

NO, use "regular" for work with that adapter. But, also, share your command, the one produce the output in your attached picture

hzeller, 2021-03-26:
The electrodragon is similar to the adapter in this project. If you have a 64 pixel high panel, then you need to set a jumper; they have a similar arrangement like in [the rpi-rgb-matrix adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3#choose-e-line-for-64x64-panels-with-132-multiplexing)