hzeller/rpi-rgb-led-matrix #1285: Full brightness flickering

rijotech, 2021-04-01:
I really thought I had something special, but after testing my project behind a business glass with the sun out, I noticed the brightness on my screen was too low. After reading a few previous posts, I thought it was a power issue, but I added an extra power supply having a total of 5v and 100a on 8 p2.5 128x64 led modules and all I'm getting is a lot of brightness pulsing.
and I'm wondering if there is a solution for full brightness options, Just in case I already have 25v  3300uf capacitors.
I need your help, Mr @Zeller.

https://user-images.githubusercontent.com/17016357/113362303-107e1d80-931c-11eb-9ae0-481419464d80.mp4

[image]

https://user-images.githubusercontent.com/17016357/113362334-24c21a80-931c-11eb-9802-3ad7ee029317.mp4

hzeller, 2021-04-02:
Did you divide the power supplies so that one only sends 5V to half of the panels and the other to the other half ?

In general, if you want brightness, you want the so-called 'outdoor' panels, so the kind that have 1:8 multiplexing.

rijotech, 2021-04-02:
Nope, I connected both power supplies to the same metal bar, I'll try dividing them.  
1:8 multiplexing? I don't really know how to all I get is a bunch of thin lines when I add multiplexing.

> Did you divide the power supplies so that one only sends 5V to half of the panels and the other to the other half ?
> 
> In general, if you want brightness, you want the so-called 'outdoor' panels, so the kind that have 1:8 multiplexing.

hzeller, 2021-04-02:
If you have not divided them, the power-supplies will fight each other and pulsing and even full switch off/on is to be expected.

If you have outdoor panels with 1:8 multiplexing, then the appropriate multiplexing option will work for them. They are brighter and typically also allow for a faster refresh rate.

rijotech, 2021-04-02:
Yep, you're right, the real problem is these are indoor modules, they are not as bright as the outdoor panels, I'll have to get and test this with outdoor modules. Thanks, Mr @hzeller

closed 2021-04-02 by rijotech