hzeller/rpi-rgb-led-matrix #1287: No module named (image / core)

Gamereric21, 2021-04-07:
Hi!  Id like to start off by saying im a beginner, and from looking through other issues, im not quite certain exactly what they were doing / supposed to do (in fairness they were in college for computer science)

Ive followed a few tutorials, on of them being a nextbus project from adafruit, one of them being an NBA scoreboard, and one an NHL scoreboard, and no matter what project I do I always get a no module named --- error.

I was trying to run a board test code in rgb-led-matrix (NBA scoreboard), and this is the error message I got

```
@raspberrypi:~/NBA-scoreboard/rpi-rgb-led-matrix/bindings/python/samples $ sudo python3 graphics.py
Traceback (most recent call last):
  File "graphics.py", line 2, in <module>
    from samplebase import SampleBase
  File "/home/pi/NBA-scoreboard/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/home/pi/NBA-scoreboard/rpi-rgb-led-matrix/bindings/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ModuleNotFoundError: No module named 'rgbmatrix.core'

```
This is the error message I got when I tried to run the nextbus transit timer project from adafruit:

```
Traceback (most recent call last):
File "nextbus-matrix.py", line 5, in <module>
import Image
ImportError: No module named Image
```

(I have a-lot more code from the first one as it was yesterday.)

I have spoken to the following, and no one seems to know why I get this error message:

Adafruit support- mike (forums)

u/bder222 (developer of MLB scoreboard on GitHub) - I dm'ed him, and we tried a little bit of troubleshooting code.  That error was after trying to run a preloaded python board test.  He said that he's tried it on new boards, and it always works fine, so im kinda lost. 

Hardware info:

- Raspberry pi zero WH
- Adafruit 16x32 dot matrix display 6mm pitch (2, currently only using one)
- Adafruit dot matrix bonnet
- 5v 4a switching power supply

(From what I can tell the nextbus installs an older version of this library, however I have tried installing the new one, and I know that the NBA scoreboard installs a new version)

(Some code / comments copied from my post on reddit)

If anyone here could help, or has any experience with this kind of error, that would be much appreciated

Thanks

xuniuer, 2021-04-13:
Maybe you should install pillow package for Image

````pip install pillow``

Gamereric21, 2021-04-13:
I can definitely try that.  I tried it on the Nextbus before, but ill see what I can do with the NBA scoreboard!

Gamereric21, 2021-04-13:
> Maybe you should install pillow package for Image
> 
> ````pip install pillow``

I tried that!  still getting that error for some reason.  Know anything else that may help?

xuniuer, 2021-04-14:
Some points to check:
1. Python3/pip3 is available by default? If not, please use ```pip3 install pillow``` 
2. Could use original project to make python binding, then ```make install``` under bindings/python directory to install python extension to your system
3. Start a python shell, try to ```import rgbmatrix``` to validate