hzeller/rpi-rgb-led-matrix #1289: Help: Qiangli outdoor, P8, full color, HUB75C, ICN2038 driver

xuniuer, 2021-04-13:
Hello,

I got some Qiangli P8 outdoor panels (model: S8C4V4, w:32,h:16 pixel) with Hub75C and ICN2038 chip. I use electrondragon panel to wire panel to raspberry pi 3B. 

According to readme, I set ```--led-row-addr-type=2```  and ```--led-multiplexing=4``` to run:

```sudo ./demo --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=4 --led-slowdown-gpio=2 --led-row-addr-type=2 -D 8 ```

I find there is a dark row each 3 rows on the panel. Please take a look the picture:
[image]

Plus: the Hub75C layout as referred: https://github.com/hzeller/rpi-rgb-led-matrix/issues/808#issuecomment-489424993

I dig the repo and find some same topics about icn2038s posted since 2018, such as #446, #976. No luck to solve the issue.

And I try to patch ```framebuffer.cc``` to add ```if (col == columns_-4) io->SetBits(h.strobe);``` as described in #446
https://github.com/hzeller/rpi-rgb-led-matrix/issues/466#issuecomment-370287035 or
https://github.com/hzeller/rpi-rgb-led-matrix/issues/466#issuecomment-372053749

These dark lines are still dead. 

Any idea? Thanks.

eliotmartin, 2021-04-13:
I had a similar issue and it turned out to be hardware related.  The connection between the matrix, the hat, and the Pi wasn't great.  After reseating everything it was resolved.

Looking at your picture I would push that left-hand ribbon cable into the socket fully (and push together any other connections that may not be fully engaged)

xuniuer, 2021-04-14:
Thanks. I check the wiring carefully, connections are full now. 
But the problem is still there even if I replace a new Pi.

xuniuer, 2021-04-14:
In addition, when I try to use ```--led-row-addr-type=0```, the 4th, 8th, 16th lines are on but with less brightness and there are some noisy colors.
[image]

xuniuer, 2021-04-23:
Interesting! @eliotmartin 

When I wire `cs` to `gnd`, the problem disappears. Take a look.

[image]

Where it is called Hub75C like this
[image]

And now just use multiplexing = 4, default row-addr-type, everything is OK.

closed 2021-04-23 by xuniuer