hzeller/rpi-rgb-led-matrix #1290: Possible to use with rpi_ws281x?

simon-hall, 2021-04-16:
Is it possible to use this marvelous library alongside https://github.com/richardghirst/rpi_ws281x ?

  * I've tried using GPIO12 for my Neopixel LEDs, but get very jittery/strobing results.
  * Using GPIO18 results in no Neopixels showing :(
  * I'm _not_ using the PWM GPIO18 -> GPIO4.

From what I can understand, this might not be a possibility due to PWM something something?

hzeller, 2021-04-16:
In general, you can remap some of the GPIOs if needed and also skip using PWM in this library (which is not recommended).

But using two libraries (this rpi-rgb-led-matrix and others such a https://github.com/richardghirst/rpi_ws281x or http://spixels.org)
that _both_ want access the GPIO in parallel will bring you into trouble because they want to access a single resource that can't be broken up easily so probably the answer is _no_.

hzeller, 2021-04-16:
If you want to free up or remap GPIO pins used by this library, here is where you would start: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c

simon-hall, 2021-04-16:
Wow, thanks for the quick answer!  I guess I'll take the 'easy' option and drive my Neopixels from a separate Pi.

Keep up the good work :)

closed 2021-04-18 by hzeller