hzeller/rpi-rgb-led-matrix #1291: P2.5 SMD-2 128x64 not working

rijotech, 2021-04-21:
I'm not really sure if this module is compatible, I purchase some outdoor P2.5 SMD-2 128x64, and nothing I have tried worked. Any Idea on how to make it work?

[image]
[image]

[image]

hzeller, 2021-04-21:
So what had you tried ? Does it now show anything or just garbled ? How does the connector look like ? Are you using level converters ? Do you know the chipset they are using ?

You need to provide all relevant information to get started.

rijotech, 2021-04-21:
I'm using the regular data cable and the active 3 board, is not showing anything the chip this modules is using is called chipone 2153, also I'm using a 5v 20a power.

So far I have tried different data cables and power supply and still not showing anything, the lights doesn't even turn on.

rijotech, 2021-04-21:
I'll try the patch mentioned on this post to see if it work.
https://github.com/hzeller/rpi-rgb-led-matrix/issues/466#issuecomment-372053749