hzeller/rpi-rgb-led-matrix #1294: LED matrix (P3(2121)64x64-32S-M1) does not show the correct output.

Yoshinese, 2021-04-24:
Hi! I have recently started working with LED matrix board and I bought a 64x64 led matrix board with the product number - P3(2121)64x64-32S-M3.
[image]
[image]

I am using the Matrix Bonet that Adafruit supplies, but quickly realized that a lot of the things do not work.  I saw that you had to solder the E and 8 Line together, and have tried to find online forums for help but without any progress. Please help I am extremely stuck.

Yoshinese, 2021-04-24:
@hzeller please help

hzeller, 2021-04-24:
If you don't feel comfortable doing the [Adafruit 64x64 modification](https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet), I suggest to get one of the [adapters](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter), for instance the [electrodragon](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/) is ready to go.

Yoshinese, 2021-04-24:
Thank you for replaying @hzeller I tried the modification but the results still stayed the same. I think it has to do with the fact that we cannot scratch the pi onto the ground plane because of the solder that is present in the Bonet. Unfortunately I looked into buying the electrodragon but it doesn't seem like it will be coming anytime soon. One thing that did changed was that the squares became much more clearer after the modification - perhaps this is going in the right direction?

Yoshinese, 2021-04-24:
Quick question what does scratch the connection from the ground pin exactly mean? Are we able to just not connect that pin 8 back to the bonnet and just directly connect it to pin 24?

hzeller, 2021-04-24:
scratch means: disconnect, otherwise you short the wire to ground and that of course does not work. So the copper trace there needs to be cut with a sharp knife until it is disconnected. If you zoom into the picture, you see what it means.

kiwidave72, 2021-05-15:
@hzeller please can you help

im getting odd display...ive had this working with out doing the mod (see sonic gif)

but now when i run things (after it sitting a week) im getting rubbish displayed.  (demo running) 

https://user-images.githubusercontent.com/8468461/118345417-ad68e300-b577-11eb-9339-9bf582b999f0.MOV

https://user-images.githubusercontent.com/8468461/118345374-6844b100-b577-11eb-9a54-3379fe837292.MOV

MasterChief1335, 2024-10-20:
Hi by me the colors are blue, purple, white 😭
Pins
R1 G1
B1 GND
N N
N GND
A B
C NC
CLK LAT
OE GND
[image]

davemaster, 2024-10-21:
Greetings, 

Try TO USE NOT A HAT, directly to your panel, (where are your panel HUB75 PORT?) TAKE A PICTURE and share your panel data connection

MasterChief1335, 2024-10-21:
> Greetings,
> 
> Try TO USE NOT A HAT, directly to your panel, (where are your panel HUB75 PORT?) TAKE A PICTURE and share your panel data connection

I worked according to the tutorial
https://www.instructables.com/RGB-LED-Matrix-With-an-ESP8266/

davemaster, 2024-10-22:
> > Greetings,
> > Try TO USE NOT A HAT, directly to your panel, (where are your panel HUB75 PORT?) TAKE A PICTURE and share your panel data connection
> 
> I worked according to the tutorial https://www.instructables.com/RGB-LED-Matrix-With-an-ESP8266/

That instructable, is for a ESP8266 based dev board, NOT for a Raspberry Pi 1-4 board.... this is not for ESP8266, so, You must deep a little for find your stuff... keep going.

MasterChief1335, 2024-10-22:
1 I never mentioned that I use Pi?
2 it works, but the colors are wrong!
3 You have to dig a little to find your stuff... keep going. NICE 🙄🤦‍♀️ I would never have thought of that

MasterChief1335, 2024-10-22:
> 1 I never mentioned that I use Pi? 2 it works, but the colors are wrong! 3 You have to dig a little to find your stuff... keep going. NICE 🙄🤦‍♀️ I would never have thought of that

[image]

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828