hzeller/rpi-rgb-led-matrix #1295: trying to run a 4 x 64x64 in a square setup

MythicalForce, 2021-04-24:
## Command Running
**sudo ./demo -D0 --led-rows=64 --led-cols=64 --led-chain=4 --led-slowdown-gpio=2 --led-multiplexing=0**

## Picture
https://cdn.discordapp.com/attachments/538149593246859313/835624086352822322/image0.jpg

## Matrix Setup
**3 - 2
1 - 0 <---- adfruit-hat connects here**

## Note
in the terminal i can see that it thinks it's a "256x64".... and not "256x256"

hzeller, 2021-04-24:
You probably want to enable the u-mapping if you have that layout
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates

MythicalForce, 2021-04-24:
> You probably want to enable the u-mapping if you have that layout
> https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates

so i added **--led-pixel-mapper="U-mapper"** and i go this

https://cdn.discordapp.com/attachments/768907118589247608/835638267742781511/image0.jpg

MythicalForce, 2021-04-24:
so i flipped the bottom 2 boards arround and now it works all that remains is to figure out why there are these lines

https://streamable.com/u1nccq

MythicalForce, 2021-04-24:
just another question!

i wanted to make this work https://learn.adafruit.com/raspberry-pi-led-matrix-display/software but sadly it states that this does NOT work with Pi4 that im using :(

hzeller, 2021-04-24:
This is because adafruit uses an ancient version of my library. Just use this one and you're good.

if you have four panels with 64x64 in a square, you get 128x128 ... 256x256 would be 16 panels.

hzeller, 2021-04-25:
For the double line situation: did you play with the `--led-row-addr-type` to see if one of the settings works ?

MythicalForce, 2021-04-25:
> This is because adafruit uses an ancient version of my library. Just use this one and you're good.
> 
> if you have four panels with 64x64 in a square, you get 128x128 ... 256x256 would be 16 panels.

huh? use the current version of **rpi-rgb-led-matrix** ? but i would need to write my own code to mirror the desktop on the matrix display?

i want to display video/youtube on the matrix

hzeller, 2021-04-25:
the Adafruit framebuffer thing uses my library, but they link in a three year old version that doesn't support this binary yet.
If you upgrade that submodule to the latest version it might work, though you might need to do some adaptions.

If you just want to watch videos, it is probably easier to use the video viewer in this project https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer

MythicalForce, 2021-04-25:
> For the double line situation: did you play with the `--led-row-addr-type` to see if one of the settings works ?

teste this now and the closest is 0 :S

davemaster, 2021-04-25:
> 
> 
> just another question!
> 
> i wanted to make this work https://learn.adafruit.com/raspberry-pi-led-matrix-display/software but sadly it states that this does NOT work with Pi4 that im using :(

You need to update (yourself) the [rpi-fb-matrix](https://github.com/adafruit/rpi-fb-matrix) library from github. That library uses an old rpi-rgb-led-matrix library version. To do this, search some function calls, just read the error messages to do so.

Best regards...

MythicalForce, 2021-04-25:
> You need to update (yourself) the [rpi-fb-matrix](https://github.com/adafruit/rpi-fb-matrix) library from github. That library uses an old rpi-rgb-led-matrix library version. To do this, search some [...]

updated it but it is still not working :/

MythicalForce, 2021-04-25:
when i try and **make** i get this

https://cdn.discordapp.com/attachments/768907118589247608/835830058026926080/image0.jpg

MythicalForce, 2021-04-25:
i've done everything correct on the board aswell
https://cdn.discordapp.com/attachments/768907118589247608/835883293206511616/image0.jpg

davemaster, 2021-04-25:
> 
> 
> > > just another question!
> > > i wanted to make this work https://learn.adafruit.com/raspberry-pi-led-matrix-display/software but sadly it states that this does NOT work with Pi4 that im using :(
> > 
> > 
> > You need to update (yourself) the [rpi-fb-matrix](https://github.com/adafruit/rpi-fb-matrix) library from github. That library uses an old rpi-rgb-led-matrix library version. To do this, search some function calls, just read the error messages to do so.
> > Best regards...
> 
> updated it but it is still not working :/
> ## Video
> 
> https://streamable.com/lnotpg
Greetings,

I did it a few months ago, I'll try it again this week, taking dust out my rasp-4 2GB, I will share my mods with all here to continue the free-common-source-coding spirit....

best regards

davemaster, 2021-04-25:
> updated it but it is still not working :/
> ## Video
>
> https://streamable.com/lnotpg

I just saw your video, but in it I can't see the command used, share it please. Also, your hardware, what RPi are you using? 

Thanks in advance

davemaster, 2021-04-25:
in this cases, the best way to start is trying one by one with those modules: works for ONE, then, WORKS for all. Share also your configuration file... It will be great

MythicalForce, 2021-04-25:
> I just saw your video, but in it I can't see the command used, share it please. Also, your hardware, what RPi are you using?
>
> Thanks in advance

## Command Line
sudo examples-api-use/demo -D4 --led-rows=64 --led-cols=64 --led-chain=1 --led-slowdown-gpio=2 --led-pixel-mapper="U-mapper" --led-gpio-mapping="adafruit-hat"

## Hardware
Raspberry Pi 4 4Gb
Adafruit RGB Matrix + Real Time Clock HAT for Raspberry Pi

davemaster, 2021-04-25:
Basic adafruit examples using their Adafruit RGB Matrix HAT works? Picture
of one of them please?

davemaster, 2021-04-25:
your configuration file?

MythicalForce, 2021-04-25:
> […](#)
> -- David Elias Flores Escalante TeleTracking SAC

what config file? the /boot/config.cfg or?

davemaster, 2021-04-26:
For use the rpi-fb-matrix You must create a configuration file *.cfg 
I running sucessfully that library with my RPi 4 2GB, 

NEXT TIME, TRY to READ all the article, here a [configuration/setup file for rpi-fb-matrix ](https://github.com/adafruit/rpi-fb-matrix/blob/master/matrix.cfg)

Best regards

davemaster, 2021-04-26:
I will explain when I re-edit original files of rpi-fb-matrix to work with new rpi-rgb-led-matrix library

gddeen, 2021-04-26:
> > […](#)
> > -- David Elias Flores Escalante TeleTracking SAC
> 
> what config file? the /boot/config.cfg or?

I have a 4 chain 64x64 horizontal but use --led-gpio-mapping=adafruit-hat-pwm

davemaster, 2021-04-26:
OK,

Divide and conquer- Had You tried your adafruit HAT with only one panel? If
it is a 64x64 need some MOD. Then, try with 2, then 3 panels. The thing is,
I use the dragon active board....

try.... only ONE PANEL, and share picture

MythicalForce, 2021-04-26:
> I will explain when I re-edit original files of rpi-fb-matrix to work with new rpi-rgb-led-matrix library
> 
> Apparently, no need to configuration file for rpi-fb-matrix
> Here an example https://github.com/adafruit/rpi-fb-matrix/blob/master/matrix.cfg

realy you got it working directly from the adafruit guide?.... on a Rpi4?

but you have a dragon board and no the adafruit hat?

davemaster, 2021-04-26:
A couple of years ago. Works perfectly (on RPi 3) but with some flickering.
My screen are 4 32x32 panels from sparkfun.com (1/16 identical that those
in Adafruit)

I follow step by step the adafruit learning guide, there was a matrix.cfg
file to define your screen configuration.

Today I will receive new microsd memories, then I will repeat that "recipe"
using my RPi 4 - 2GB, and my screen 64x64 (arrangement 2x2 panels)

Best regards

fachpost60, 2021-06-10:
Hi! Did you get it working with the current version of rpi-rgb-led-matrix? I tried it, but get this error messages:
[image]

closed 2025-11-27 by marcmerlin