hzeller/rpi-rgb-led-matrix #1296: 128x64 panel using ICN2038S chip

WernerFettke, 2021-04-27:
Hi, 

I purchased some 128x64 panels from led-card.com (https://www.led-card.com/p2-fullcolor-led-indoor-module.html), the panels are P2 and from what I can see they using the following chip ICN2038S.

I've tried setting the panels up with the Adafruit hat, as well as the ElectroDragon Pi shields version 1 and 2, the best I can get it to display is as per the image below, it almost looks like the mapping is out, I've tried various mapping options, row address types, etc etc, but still no luck.

The panels have no E or D pins (I've also tried setting up the various cards as if there was an E pin), but no luck.

[image]

I also chatted to the guys at led-card.com and they suggest rotating the imageby 45 degrees, but this option is not available under the rotation options, the value needs to be in 90 degree increments.

The attached image is what I get when I run the wiring directly off the Pi, I used the Smiley Face wiring option, without using the E or D pins as there are no E / D pins on the Hub 75 connector.

[image]

Any assistance would really be appreciated as I've purchased 12 of these panels and would really like to get them working on my Pi.  (I've tried on a Pi2 / 3 and 4), also tried using different builds as suggested.

Many Thanks in advance.

Regards

Werner

hzeller, 2021-04-27:
Did you go through all `--led-multiplexing` and `--led-row-addr-type` settings ? The fact that there are only three address pins A, B, C indicates that these options are very relevant. Maybe they have a special mapping ?
Did you get a datasheet for the Matrix and their pixel-mapping, maybe you need to add a new one.

WernerFettke, 2021-04-27:
Many thanks for getting back to me so quickly ! I have asked the guys at led-card.com for datasheet, but unfortunately they have not responded as yet. All I can see under the specifications of the panels is the following, which I know does not help much.

Pixel pitch | 2mm
-- | --
Module Size | 256X128 mm
Model Pixel | 128X64Pixels
lattice density | 250000 dots/square meter
Pixel form | 1R1G1B SMD1515
Best viewing distance | 1m~20 m
Max power consumption | 2000W/square meter
Ave power consumption | 600W/square meter
Max power per module | 25W
warranty | 1 year
Drive mode( Optional) | Constant current 1/32scan
Control system | DVI card+Full color control card
Pixel type | Real pixel
refresh frequency | >=400HZ
Horizontal View Angel | > 110 degree
Working Environment | Temp:-20Celsius degree-+60Celsius degreeHumidity : 10%-90%
Working voltage | DC 5V
Lifetime | >=100000 hours
Gray scale | 12 bit/ 1 color
Color display | manual adjust
Brightness | >600 cd/square meter
Blind spot rate | <0.00001
Waterproof Class | IP43
Software | Support window series system

In terms of trying the multiplexing and row address options, I have been through them all, even going as far as setting the row address option and then trying each multiplexing options, then trying the next row address and going through the multiplexing options again, until I have been through all the row address and multiplexing options.

As soon as I hear back from them I will send you an updated.

Thanks for the great software, and for being so pro-active when it comes to resolving issies.

Regards

Werner

WernerFettke, 2021-04-27:
Just to provide some more feedback, if I try the various -led-row-addr-type options, The panel only displays something when I use options 0 and 4.
Option 0 gives the output as per my original post, and option 4 results in the attached image.
[image]

WernerFettke, 2021-04-27:
Hi, Apparently they don't really have anything, however I've been able to find the following, apparently the panels I'm using are using the ICN2038S chip, I could only find the following:

[image]
[image]

I was also able to find some info on the non S version ICN2038, however I believe the S version is the replacement for the non S version:

[image]
[image]

WernerFettke, 2021-04-27:
Hi, After playing with the ABC pins and the settings, I was able to get the panel to work as intended, the settings I used was: sudo ./demo -D 1 runtext.ppm --led-no-hardware-pulse  --led-slowdown-gpio=2 --led-rows=64 --led-cols=128 --led-multiplexing=0 --led-row-addr-type=3

Thank you for your help and awesome software !!!

closed 2021-04-27 by WernerFettke

Overboard-code, 2022-04-23:
I have the 128x64 1/32 panel and an ElectroDragon  Can you tell me what you did on the driver board?  Did you jumper 4 or 8 to E?

WernerFettke, 2022-04-24:
Hi,

For the panels I was using I did't need to do anything on the ElectroDragon card,  my issue was resolved by using row-address-type option 3.... have you tried running the demo with the same options I used in the previous post where I got it to work ?

Overboard-code, 2022-04-27:
My board does have an E pin  
[image]

So I had to use the jumpers  Different.  no standards yet  It's still early days I guess