hzeller/rpi-rgb-led-matrix #1297: Thank you for the awesome project. I searched to see if there was a PHP library, but couldn't find any. Has anyone run across one?

thomastthai, 2021-04-28:


hzeller, 2021-04-28:
There is a [C-binding](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/led-matrix-c.h) provided in this project which should make it simple to bind to PHP.

However a word of warning since PHP is often used in web applications: you absolutely want to make sure to drop privileges when running this in a PHP process - you don't want to have a process running as root wired up to a language with a known large security attack surface connected to the network.

Instead of that, I suggest to run a server with the rpi-rgb-led-matrix running receiving just image content over the network. I usually use the [FlaschenTaschen protocol](https://github.com/hzeller/flaschen-taschen/blob/master/doc/protocols.md) (which receives simple PPM images over UDP) for which it is [simple to compile a server](https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display) that uses this led matrix.

thomastthai, 2021-04-28:
Thank you for your response! The server solution works!

closed 2021-04-29 by hzeller