hzeller/rpi-rgb-led-matrix #1298: Error: Running make on Python binding

AWSb24, 2021-04-30:
I tried to run make according to the readme on the python binding. It results in several warnings and one error. I am running python3.9 on an RPI4.

`rgbmatrix/graphics.cpp:3442:41: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
 3442 |   __pyx_type_9rgbmatrix_8graphics_Color.tp_print = 0;
      |                                         ^~~~~~~~
rgbmatrix/graphics.cpp:3450:40: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
 3450 |   __pyx_type_9rgbmatrix_8graphics_Font.tp_print = 0;
      | `

How can I solve this?

rhtenhove, 2021-05-22:
This is because tp_print was removed in python3.9 (and only deprecated in 3.8). You can ignore the warnings.

You can solve this on a raspberry pi using pyenv to install python3.8. The following is based on my shell history on a raspberry pi 3 running ubuntu 21.04 64-bit. If something isn't working, let me know.

We're adding things to root, since root needs to run the library to work properly.

```shell
curl https://pyenv.run | sudo bash
```

Add the following to `/root/.bashrc`

```shell
export PYENV_ROOT="$HOME/.pyenv"
export PATH="$PYENV_ROOT/bin:$PATH"
eval "$(pyenv init --path)"
```

Finally run the following in a root shell `sudo su`. `cd` to the root of the rgb matrix git repo for the make commands.

```shell
apt install -y libbz2-dev libreadline-dev libssl-dev
pyenv install 3.8.10
pyenv global 3.8.10

pip install Pillow

make build-python PYTHON=$(which python3.8)
make install-python PYTHON=$(which python3.8)
```

Then you can run things like

```shell
/home/ubuntu/rpi-rgb-led-matrix/bindings/python/samples/runtext.py --led-gpio-mapping=adafruit-hat --text="Hello, world!"
```

nvski, 2021-05-27:
These `.cpp` files were compiled by old cython. If you don't want to downgrade to python 3.8, you can to recompile them with newer cython:

```
python3 -m pip install cython
cd path_to_repo/bindings/python/rgbmatrix/
python3 -m cython -3 --cplus *.pyx
```

mikewebkist, 2021-06-14:
It took a slight variation on @nvski's comment for me to succeed with Python 3.9.5. I needed to use `language_level=2` instead of 2:
```
python3 -m cython -2 --cplus *.pyx
```

sityware, 2021-11-11:
Seems like it can be fixed in code.

rename tp_print to tp_vectorcall_offset
in 
/bindings/python/rgbmatrix/core.cpp
/bindings/python/rgbmatrix/graphics.cpp

josh1hart, 2021-11-12:
@sityware You saying renaming every mention of tp_print in each line to tp vectorcall_offset? for both core and grahpics? nothing else on top of this?

stefffe, 2021-11-19:
That seems to fix the issue. I just did a search and replace for tp_print to tp_vectorcall_offset in core.cpp and graphics.cpp, and after that I can build and successfully run the python samples with Python 3.9.2.

hzeller, 2021-12-03:
I've updated now updated the pre-generated cpp files with the latest cython; can you test if this is now working for you ?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828