hzeller/rpi-rgb-led-matrix #1300: Works Then Stops Working

WernerFettke, 2021-05-02:
Hi, 

I had a issue last week with my 128x64 panels, After your suggestion of focusing on the ABC connectors, (The panels I have are only ABC, there are no D or E pins). I was able to get the panel to work, it was originally on a Pi 3, but there was quite a bit or flickering, so I upgraded to a Pi and the display is now rock soliday working with 1 panel...... the panels are P2, so it looks beautiful.

I was able to get the panel to work as intended by using --led-row-addr-type=3

Thanks for your assistance with this.

I have decided to expand on the project and add a couple more panels to mix..... the display is now made up of 4 x 128x64 panels across, and 2 rows of these . . . in other words 8 panels in 2 rows of 4, giving me a total dsisplay size of 512 x 128.

I have played with the various setting, eg GIOP Slow Down, row address type, pwm-lsb-nanospeed, multiplexing, etx but Im having the following issue..... when it starts up the image is not prefect, but it's displaying something, however after a couple of seconds the image goes pear shaped, and sometimes it comes back for a second, but the disappears again...... I have checked that nothing else is using the GPIO's (everything there is disabled).

It's just weird that there is something, but after w few seconds it gets scrambled, as per the video below:

https://user-images.githubusercontent.com/53822201/116811853-b99e7a80-ab4b-11eb-92fb-3f25daaba969.mov

The black lines in the video are just the recording, the image is complete, just the camera recording at a different ffrequency.

Any suggestions ?

Regards

Werner

WernerFettke, 2021-05-02:
I forgot to mention, I have wired the pins directly to the Pi for each row..... and I'm using P2 as per the image below:

[image]

closed 2021-05-02 by WernerFettke

davemaster, 2021-05-02:
Looks like a power supply issue