hzeller/rpi-rgb-led-matrix #1301: One 64x64 Matrix

infinite0007, 2021-05-02:
What are the exactly config settings if i only have one matrix 64x64?
Keep saying that 4 panels is needed. How can I change that?

hzeller, 2021-05-02:
This depends on your panel. Follow the instructions in the [README](https://github.com/hzeller/rpi-rgb-led-matrix#types-of-displays) to find the settings that work for your panel.

davemaster, 2021-05-03:
> 
> 
> What are the exactly config settings if i only have one matrix 64x64?
> Keep saying that 4 panels is needed. How can I change that?

in default, the size of panel are 32x32 and from adafruit (1/16 scan), You must know what kind of panel do You have, and its features... then, if it is supported by this library...

Best regards

infinite0007, 2021-05-05:
forgot that if you choose the quality setting, that you need to wire SOL4 with SOL8 that was the issue.

closed 2021-05-05 by infinite0007