hzeller/rpi-rgb-led-matrix #1302: Close shown Image

Fipschen, 2021-05-08:
Hey,

First of all: Thanks for this lib!
As next: sorry, my english isn't very good, but I'll try my best.

My problem is, that is can't close the Image, which is displayed.
Every second I'm generating a new Image with the same file name, but I can't save it, cause "Permission denied". I'm not sure, if I know why it's denied, but I think, the lib takes the file open while it's displaying it and so i can't change it.

Is there a possibility to close the Image-file and display the new one?

Thank you very much.

P.S.: I'm using Python

hzeller, 2021-05-08:
The library changes the permissions from 'root' to 'nobody', for security reasons: after the hardware set-up is done (which needs to be done as root), it switches to a user that has less privileges, so that if there is something going wrong in the program, it won't have root privileges thus contains potential damage because it can't owerwrite system files.

Or a file owned by a different users. Now, if you stored your image initially as 'pi' user, it will not be able to overwrite it as user 'nobody'.

I suggest to `chmod a+w <filename>` on the image beforehand (change permissions that 'all' can 'write'), so that you then can overwrite it as a different user.

(you can also us the `--led-no-drop-privs` option to stay root, but that is of course a less secure).

Fipschen, 2021-05-08:
Thanks for that information.

I activated the option ---led-no-drop-privs and now it works. That's fine for me.

I tried some things with chmod, but nothing of this didn't work.

Thank you :)

closed 2021-05-08 by Fipschen