hzeller/rpi-rgb-led-matrix #1303: Upper half yellow instead of white

Fipschen, 2021-05-09:
Hey together,

I've a small problem with the colors of my Panel:
I've this Board: https://joy-it.net/de/products/RB-MatrixCtrl to connect my Pi4 with the Panels.
The upper half of the panels (only one parallel) shows yellow instead of white.
I tried to connect the Panel one of the other Ports on the controller and on the secound was the upper half white.
So there a wiring-issue on the controller? Could I correct it in the Software by changing the GPIOs?
Or what could I do?

Thank you very much and have a nice day

davemaster, 2021-05-09:
Greetings

Check again the manual for connections. Also, share what type of panel do you have.

Fipschen, 2021-05-09:
If there is no fail on the board, the wiring is correct.
I'll check it now fast.

On the backside of the Panels is written: "P3-64x64-32S-2012-18G-1.2", next Line: "QD"

Fipschen, 2021-05-09:
Can't check the wiring over the control board cause of the level shifter. So i only have to hope, they're correct connected.

Fipschen, 2021-05-09:
I have checked the lib/hardware-mapping.c with the documentation of my panel (from the seller; https://joy-it.net/files/files/Produkte/LED-Matrix01/RB-MatrixCtrl-Anleitung-20201022.pdf) and there is no mapping matched.

Could I change the mapping of the GPIOs in the hard-mapping.c without destroy something?

davemaster, 2021-05-09:
well, now, if your power supply is "weak", the colors looks different....
check your panel(s) power requirement

Fipschen, 2021-05-09:
My power supply have 5.09 V and  40 A. If I get the Voltage up and down, the colors don't change, only the brightness.

davemaster, 2021-05-09:
Had you tried with another panel? Anyone? Share a picture of your system

Fipschen, 2021-05-09:
I have to panels in one chain. I've changed there position, but the same problem. I only have this panels at the moment.

[image]

davemaster, 2021-05-09:
ok, also a picture of the failure...thanks

Fipschen, 2021-05-09:
As a example.
With other pictures or text or Pixels the same.

[image]

hzeller, 2021-05-09:
If you know that your level converter board maps the blue line of that chain that is not working to a different GPIO, you can change it in the corresponding [hardware-mapping.c](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c). So for instance, if it is the problem in the first chain, you have to change the value `.p0_b1` for the hardware mapping you are using (default is `"regular"`).

Fipschen, 2021-05-09:
I've tried this with the documentation of the seller, but there was no changing (not better, not worser)

hzeller, 2021-05-09:
did you re-compile, and, if you are using Python, re-build and re-install the Python libraries ?

Fipschen, 2021-05-09:
I'm not safe with C and only a little with Python.
Think, Python re-build with
`make build-python PYTHON=$(which python3)
sudo make install-python PYTHON=$(which python3)`
works.
Could you help me please to re-compile C? 
Thanks you very much.

hzeller, 2021-05-09:
if you have 'make clean' and re-build Python and install it, the Makefile will take care of re-compiling the C++ library automatically.

Fipschen, 2021-05-09:
Okay, I re-compiled and re-build all, but still the same Issue.
But I'm not sure, if the documentation is correct, because most of the GPIOs in the mapping don't match the doc.
I'll try other Pins, maybe I'll find the right one.

davemaster, 2021-05-09:
change 

> 
> 
> As a example.
> With other pictures or text or Pixels the same.
> 
> [image]

how much panels are in this picture, 1 or two, if are two, change order of panels... panel issue may be

Fipschen, 2021-05-09:
> change
> 
> > As a example.
> > With other pictures or text or Pixels the same.
> > [image]
> 
> how much panels are in this picture, 1 or two, if are two, change order of panels... panel issue may be

They are two panels, order I already have changed, but the same.

Fipschen, 2021-05-09:
Okay... now, I don't know how to go on.
I changed the .p0_b1 to all Pins from the other Panels and did after saving of hardware-mapping.c this:
`make clean
make build-python PYTHON=$(which python3)
sudo make install-python PYTHON=$(which python3)`
and started my script after that.

But with all Pins is the same issue.

hzeller, 2021-05-09:
if you know that you output the right color on the right GPIO, then either your GPIO pin on the Pi is broken or used by some oother program. Make sure that `p0_b1` is not used by some other program (e.g. SPI on the regular mapping).

Or the wiring to/from the voltage level converter, or the level converter itself.

Fipschen, 2021-05-09:
I'm on the way to play a new image on the SD for the Pi.
There was no other program, that's the only thing, the Pi has to do.

If the Issue stays, I will change the Pi. If the Issue stays after this to, I'll contact the seller, to get a new Board to check it.

Fipschen, 2021-05-09:
Now, I think, something is damaged. The Panels don't do something anymore...
Time to connect the seller. >.<

Fipschen, 2021-05-18:
Hey together,

today the new controller board arrived and now it all works fine, so it only was a hardware issue.
Thank you very much for your support.

closed 2021-05-18 by Fipschen

davemaster, 2021-05-19:
I knew it...

best regards