hzeller/rpi-rgb-led-matrix #1304: Not really an issue, need help

spannt, 2021-05-11:
Hello, not sure where else to put this. I'm building a stargate, using 32X32 panels, a raspberry pi 3 b+ with a led matrix hat. For some reason no matter how much I try, I can't get the audio to sync with the video. I'm using video-viewer. If anyone has any suggestions, that would be awesome. I'm using dietpi 64bit version. 

[image]
[image]

spannt, 2021-05-11:
Yeah I know that, the sound is separate in a mp3. The sound is dead on to the video when played on another raspberry pi. Could it be the video isn't running as fast as it should? Thats the issue I'm having.

spannt, 2021-05-11:
Here's how I'm running the video

$MATRIXDIR/video-viewer -F --led-scan-mode=1 --led-rows=32 --led-cols=32 --led-chain=31 --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=3 --led-multiplexing=1 ./abydos_dial.mp4>/dev/null 2>&1 &

hzeller, 2021-05-11:
yes, the video decoding uses a lot of resources and might play frames back slower than they should, in prticular if the video needs to be scaled down a lot by the video-viewer.

For minimum overhead at runtime, I suggest to create a stream-file from the video, and then play that back with the image- viewer. The output file will be huge, but it means there is no overhead at runtme except reading it from disk. There is something in the [examples section of the video-viewer](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#examples-2) that shows how to do that.

hzeller, 2021-05-11:
One thing I noticed: your chain of 31 panels is somewhat long and you might experience lower refresh rate than pleasent.
(I'd consider running three parallel chains and map the video accordingly)

If you see low refresh rate (using `--led-show-refresh`), you can play around with `--led-pwm-bits` and `--led-pwm-dither-bits` to see if you can optimize a bit the refresh.

spannt, 2021-05-12:
Thank you, I will start with the images and go from there. BTW I'm a member of the The Repilica Prop Forum, I have been sharing your github with everyone. They keep asking for me to share my code, so I send them here.  :)

spannt, 2021-05-12:
I got it, :) Thanks for your help

[https://youtu.be/fGfVdd5qGIg]

hzeller, 2021-05-15:
Yay! Have fun with your Stargate :star: :door:

closed 2021-05-15 by hzeller