hzeller/rpi-rgb-led-matrix #1305: 32*256 only able to run at 32*255

tomscatbilly, 2021-05-12:
So I have a chain of 4 32x64 panels to give me 32x256. However I cant seem to get them to run at the full 256 columns, I can only do up to 255.

I get the error
`OverflowError: value too large to convert to uint8_t` whenever I try to run a 256.
I've tried displaying an image that is 32x257 but it has no effect

Any help is appreciated!

hzeller, 2021-05-15:
This is not an error that is produced by the C++ library. Are you using some sort of wrapper library ?

tomscatbilly, 2021-05-15:
i was just using the python3 library. i realised that whilst the image co-ords start at 1, the width of the dot matrix started at 0. A few tweaks later and i managed to fix it. 

The project 'streams' an image from my server that does all the generating and data handling.

`DMI_IMAGE = Image.open(IMG_STREAM.raw)
DisplayImage = Image.new("RGB",(matrix.width+1,matrix.height))
DisplayImage.paste(DMI_IMAGE)
DisplayImage.load()
matrix.SetImage(DisplayImage,-1,0)`

Probably not the best way to do it but it works so...

tomscatbilly, 2021-05-22:
I have since realised that this doesnt fix the issue.
it just "crops" the image then displays it

so if the image normally would have 0,0 on the image be mapped to 1,0 on the dot matrix
the code in my earlier comment (the "fix" that didnt), maps 1,0 to 0,0 
so either way i end up either losing the left-most column or the right-most column.