hzeller/rpi-rgb-led-matrix #1306: "V-mapper: no such mapper" error when running new layout

topazadam, 2021-05-15:
Hi,

I have managed to build a number of successful applications using this library utilising the U-mapper. When I switched layouts and required the "V-mapper:Z", I get the following error (shown below with demo 3 running);

`
 pi@pi-hole:~/matrix-screen-driver/examples-api-use $ sudo ./demo -D3 --led-rows 32 --led-cols 64 --led-pixel-mapper="V-mapper:Z"
V-mapper: no such mapper
Size: 64x32. Hardware gpio mapping: adafruit-hat
Press <CTRL-C> to exit and reset LEDs
`

hzeller, 2021-05-15:
Sounds like you are not using the latest version of this library. The V-mapper has been added in March last year.

hzeller, 2021-05-15:
Given that the program reports to use the `adafruit-hat` hardware mapping, but you haven't provided the option `--led-gpio-mapping=adafruit-hat`, I suspect you are using the fork Adafruit did a while back from my library to sell their hats, but that is ancient (they have an ancient fork where they just enable their hat by default).

Just use this official version of the library which includes the V-Mapper and use `--led-gpio-mapping=adafruit-hat` with it to work with your hardware.

And you probably want to do the little [hardware hack](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) to decrease flicker with that HAT.

(This library has nothing to do with Adafruit. They're just using my library)

topazadam, 2021-05-15:
Perfect, thanks for the prompt solution and thanks for the great work on the library @hzeller.

closed 2021-05-15 by topazadam