hzeller/rpi-rgb-led-matrix #1307: 128x64 flickering / mapping help please.

WernerFettke, 2021-05-18:
Hi,

I think we have either reached the limit of what the Pi can do or we possibly doing something wrong. Our display is current using 128x64 P2 ABC 1/32 panels....... and our layout is 2 high by 4 wide, using the first and second output from the Electrodragon controller, each output is running 4 chained panels. Our total display size is 512 x 128.

Our problem is flicker, I know the 1/32 panels are more prone to flicker than 1/8 or 1/16, however finding 128x64 panels that are not 1/32 seem to be an issue.

Depending on what content is being displayed we rarely get anything close to 100Hz on the output. We have tried almost every configuration on the command line (Basically been at this for 2 weeks).... we have tried with a Pi2 / 3 and 4, with varying results,  the power leads are the ones supplied and we using a 40A power supply to run everything.

So far our best output is around 50 to 60Hz using the following options on the video-viewer in the utils folder: 
--led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-chain=4 --led-row-addr-type=3 --led-show-refresh --led-slowdown-gpio=5 --led-parallel=2 --led-pwm-bits=7

Because the panels are ABC we are using --led-row-addr-type=3
Using anything lower on the GPIO slow down results in an un-viewable output
In an attempt to speed things up we set --led-pwm-bits=7 You can go lower which seems to help, but does not produce a quality output.

We have tried using --led-pwm-dither-bits which gives us an extra couple of frames, but seems to introduce a new issue.... a pulse every couple of seconds on a regular basis.

We have tried the python player as well as the C player in the utils folder with basically the same result.

Reducing the amount of panels from 2x4 to 2x3 does give us better performance, however it's not where we need to be.

For now we have decided to get 2x3 (384x128) working and if that is possible, then look at 2x4. 

A suggestion was made to use all 3 outputs from the Electrodragon card (in other words run 2 panels per output still giving us the same resolution but spreading the load across all 3 outputs instead of only 2 currently) which should increase performance.... hopefully.

Here is where we have hit a road block..... we need to end up with a display in a resolution of 384 x 128, with a layout as follows:
Top Row -> Panel 1 - Panel 2 - Panel 3
Bottom Row -> Panel 4 - Panel 5 - Panel 6

Using the 3 outputs from the Electrodragon card we where looking at doing the following:
Output 1 -> Panel 1 - Panel 2
Output 2 -> Panel 4 - Panel 5
Output 3 -> Panel 3 - Panel 6

Our issue is how do we map Output 3 to the ends of each row showing the ends of each row ?

Any assistance here would really be appreciated.

@hzeller your code is awesome and a big thank you for the work you have done here, many of us would still be battling along to get one panel to work if it wasn't for your work ! Thank You !

raspberry-pi-maker, 2021-07-19:
First "--led-slowdown-gpio=5"  is wrong, This value should be  ~ 4.

And it's a good idea to use 3 Hub75 cables. The longer the Daisy chain, the lower the refresh rate.
You can use VMapping for 3 hub75 cables like this.

         Vmapper                             Vmapper:Z

  [O < I] [O < I] [O < I]             [I > O] [I > O] [I > O]
   ,---^   ,---^   ,---^               ^       ^       ^
  [O < I] [O < I] [O < I]             [O < I] [O < I] [O < I]
       ^       ^       ^                   ^       ^       ^
      #1      #2       #3                 #1      #2       #3
         Pi connector (three parallel chains of len 4)

HZeller explains in detail in the following link.
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates

Steveraws, 2022-04-28:
As i see you have the same problem as me. Probably the abc collum slect is different then other pannels.

Some 128x64 use clock/serial in collum select instead of demuxers.

These ic need more clock cycles the demux and so are slower. Probably wy your refresh rate is lower

closed 2025-11-27 by marcmerlin