hzeller/rpi-rgb-led-matrix #1311: OverflowError: can't convert negative value to size_t on SetImage() call

nikhilh-20, 2021-06-05:
Hello,

I'm in the middle of a hobby project where I'm trying to light up a 32x64 LED display (with `adafruit-hat` / RPi4) through a Django web interface. I hit `can't convert negative value to size_t` when I call `SetImage()`.

Now, the code that I'm using within Django works perfectly if I take it out and refactor it to work standalone from the cmdline. So, something like `python3 project.py` will light up the LED display perfectly. It's only when I introduce that code within Django (and also if I multi-thread the code within `project.py`) do I hit the OverflowError.

Any idea how to debug this or have you come across this before? Thanks!

hzeller, 2021-06-05:
This is something in the Python to native library layer, which you have to figure out on your machine. I'd double check that you compiled and installed the Python API with the same Python version you use.

To dig into the internal details, follow the stacktrace; what is in line 46 in the core.pyx file ?

nikhilh-20, 2021-06-05:
I was able to resolve the error by calling `SetImage` with `unsafe=True`.

[L45](https://github.com/hzeller/rpi-rgb-led-matrix/blob/dfc27c15c224a92496034a39512a274744879e86/bindings/python/rgbmatrix/core.pyx#L45) in `core.pyx` seems to accessing an `image32` property of the PIL image object. Maybe it did not return an expected value? I'm not familiar with inner workings of PIL.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin