hzeller/rpi-rgb-led-matrix #1313: undefined reference to `rgb_matrix::RGBMatrix::RequestInputs(unsigned long long)' error

K0rkunc, 2021-06-07:
hi im beginner on linux i need ur help :(

integrating ur library my project but i cant work im getting some errors

How to fix that ? im using input-example can u give me sample or help plz

pi@raspberrypi:~/matrix $ gcc -o main main.cpp
/usr/bin/ld: /tmp/ccGzawJ5.o: in function `main':
main.cpp:(.text+0x4c): undefined reference to `rgb_matrix::RGBMatrix::Options::Options()'
/usr/bin/ld: main.cpp:(.text+0x88): undefined reference to `rgb_matrix::RGBMatrix::CreateFromFlags(int*, char***, rgb_matrix::RGBMatrix::Options*, rgb_matrix::RuntimeOptions*, bool)'
/usr/bin/ld: main.cpp:(.text+0xcc): undefined reference to `rgb_matrix::RGBMatrix::RequestInputs(unsigned long long)'
/usr/bin/ld: main.cpp:(.text+0x190): undefined reference to `rgb_matrix::RGBMatrix::AwaitInputChange(int)'
/usr/bin/ld: /tmp/ccGzawJ5.o: in function `__static_initialization_and_destruction_0(int, int)':
main.cpp:(.text+0x2e8): undefined reference to `rgb_matrix::RGBMatrix::Options::Options()'
/usr/bin/ld: main.cpp:(.text+0x2f0): undefined reference to `rgb_matrix::RuntimeOptions::RuntimeOptions()'
collect2: error: ld returned 1 exit status

hzeller, 2021-06-10:
The error you see is that the linker doesn't find the functions defined in the library. Reason is, that it doesn't know that it has to link the library. So you essentially just need to add `-L<path-to-the-lib-dir> -lrgbmatrix` to the compiler call (with the appropriate path for the lib dir).

In the `examples-api-use/` is a section how you can [set up your Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#integrating-in-your-own-application) to link the library.

K0rkunc, 2021-06-10:
thank u so much solved

K0rkunc, 2021-06-10:
thank u so much solved ,

closed 2021-06-10 by K0rkunc