hzeller/rpi-rgb-led-matrix #1314: Flicker problem

K0rkunc, 2021-06-11:
**hi my screen flicker how to fix that

i tryed slowdown 3 4 5  ditherbits 0 1 2 pwm nano sc 50 150 250 35 1500 i tried but it didn't work

my fps 189/193 but sometimes it turns 40 then goes back to normal again

MY Codes**

#include "led-matrix.h"
#include "./include/pixel-mapper.h"
#include "./include/content-streamer.h"
#include "./include/graphics.h"
#include "./lib/gpio.h"
#include <fcntl.h>
#include <math.h>
#include <signal.h>
#include <getopt.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <sys/stat.h>
#include <sys/time.h>
#include <sys/types.h>
#include <unistd.h>
#include <algorithm>
#include <map>
#include <string>
#include <vector>
#include <Magick++.h>
#include <magick/image.h>

using namespace rgb_matrix;
using rgb_matrix::GPIO;
using rgb_matrix::RGBMatrix;
using rgb_matrix::Canvas;
RGBMatrix::Options defaults;
rgb_matrix::RuntimeOptions runtime_opt;

volatile bool interrupt_received = false;
static void InterruptHandler(int signo) {
  interrupt_received = true;
}

int main(int argc, char *argv[]) {
  
  Magick::InitializeMagick(*argv);
    
  Color color(255, 0, 0);
  Color bg_color(0, 0, 0);
  Color flood_color(0, 0, 0);
  Color outline_color(0,0,0);
  //defaults.hardware_mapping = "regular";  // or e.g. "adafruit-hat"
  defaults.rows = 64;
  defaults.cols = 128;
  //defaults.chain_length = 1;
  defaults.pwm_dither_bits=1;
  defaults.pwm_bits=11;
  //defaults.parallel =1;
  defaults.show_refresh_rate=1;
  //runtime_opt.daemon=0;
  runtime_opt.gpio_slowdown=3;
  
  RGBMatrix *matrix = RGBMatrix::CreateFromFlags(&argc, &argv, &defaults,&runtime_opt);
  Canvas *canvas = rgb_matrix::CreateMatrixFromFlags(&argc, &argv, &defaults,&runtime_opt);

  signal(SIGTERM, InterruptHandler);
  signal(SIGINT, InterruptHandler);
  rgb_matrix::Font font;
  const char *fontFile = "./fonts/texgyre-27.bdf";
  font.LoadFont(fontFile);
 
 while (!interrupt_received) {
     
         // matrix->SetPixel(50,50,114,115,0);
           
           rgb_matrix::DrawText(canvas,font,8, 25,color, NULL, "TEOMAN");
          
  }

 

  return 0;
}

raspberry-pi-maker, 2021-07-19:
I think the code should be like this

```c
rgb_matrix::FrameCanvas *canvas;
RGBMatrix *matrix;
RGBMatrix::Options matrix_options;
rgb_matrix::RuntimeOptions runtime_opt;

defaults.pwm_dither_bits=1;
defaults.pwm_bits=11;
//defaults.parallel =1;
defaults.show_refresh_rate=1;
//runtime_opt.daemon=0;
runtime_opt.gpio_slowdown=3;

  ......

matrix = RGBMatrix::CreateFromOptions(matrix_options, runtime_opt);
canvas = matrix->CreateFrameCanvas();

 while (!interrupt_received) {
          matrix->SetPixel(50,50,114,115,0);
          canvas = matrix->SwapOnVSync(canvas);           
  }
```

K0rkunc, 2021-07-20:
> I think the code should be like this
> 
> ```c
> rgb_matrix::FrameCanvas *canvas;
> RGBMatrix *matrix;
> RGBMatrix::Options matrix_options;
> rgb_matrix::RuntimeOptions runtime_opt;
> 
> defaults.pwm_dither_bits=1;
> defaults.pwm_bits=11;
> //defaults.parallel =1;
> defaults.show_refresh_rate=1;
> //runtime_opt.daemon=0;
> runtime_opt.gpio_slowdown=3;
> 
>   ......
> 
> matrix = RGBMatrix::CreateFromOptions(matrix_options, runtime_opt);
> canvas = matrix->CreateFrameCanvas();
> 
>  while (!interrupt_received) {
>           canvas->SetPixel(50,50,114,115,0);
>           canvas = matrix->SwapOnVSync(canvas);           
>   }
> ```

thanks for your answer

There will be another small problem, I want to show a picture, I want to show another small picture or text on it, but the picture on the base is broken, how can I solve this, can you give an example about it?

raspberry-pi-maker, 2021-07-20:
Magik++ is PIL compatible graphic library and Hzeller's library supports it.
I hope this code sample might helpful

```c
Image img;

img.read("your image.jpg");
img.resize( Geometry(128,64));

//if you need another graphic job
// Do it on the img object

//draw the image on the canvas
for (size_t y = 0; y < img.rows(); ++y) {
  for (size_t x = 0; x < img.columns(); ++x) {
    const Magick::Color &c = img.pixelColor(x, y);
    canvas->SetPixel(x, y,
                        ScaleQuantumToChar(c.redQuantum()),
                        ScaleQuantumToChar(c.greenQuantum()),
                        ScaleQuantumToChar(c.blueQuantum()));
  }
}
canvas = matrix->SwapOnVSync(canvas); 
```