hzeller/rpi-rgb-led-matrix #1315: Red light is displayed instead of expected white when trying to fill 32*16 panel with maximum white.

shinych, 2021-06-23:
Hello. 
Thank you for the awesome library!

I am trying to experimentally see what is the power consumption when the panel is turned on with maximum brightness where all LEDs are on with #FFFFFF color. For some reason, the panel gets dark red instead of the expected bright white.

I am using this 32*16 panel https://de.aliexpress.com/item/32897324696.html with these options: `--led-rows=16 --led-multiplexing=3 --led-row-addr-type=2` also added `--led-slowdown-gpio=2` for Raspberry Pi 4 (as suggested in the issue 681)

This is my python function for filling the panel with the given color:

`def display_fill(self, color):
    canvas = self.matrix.CreateFrameCanvas()
    canvas.Clear()
    for y in range(0,16):
        graphics.DrawLine(canvas, 0, y, 31, y, color)
    canvas = self.matrix.SwapOnVSync(canvas)`

It works as expected when I pass any color up to approx. 243 brightness: `graphics.Color(243,243,243)` 

But when I pass e.g., `graphics.Color(255,255,255)`, the LEDs are on with a pale red color, which seems like some kind of a safe mode. (?)

I also noticed that when I reduce the number of lines in my function from 16 to 12 (so only 3/4 of the LEDs are on), then the fill with #FFFFFF also works fine.

What can be the reason for this behavior? 

Thank you.

fjansson, 2021-07-15:
Just a guess: could you be overloading the power supply? Red LEDs require less voltage to light than blue or green, so if the voltage drops the red ones would work the longest.

shinych, 2021-07-19:
@fjansson thanks for your reply! Possibly, this was the issue. 

With another power supply, it works properly now.

I am wondering if this "falling back to dark red" is by design or just a coincidence. Seems to be a neat feature for such cases!

SchlenkR, 2024-01-22:
..same issue here, and: Reducing the number of LEDs helps. I didn't try another power supply, but I'll try.
BTW: I'm using a complete different library :D So I guess you can close that issue here.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828