hzeller/rpi-rgb-led-matrix #1317: How to create a few FrameCanvases in Python?

xXZyzzXx, 2021-07-01:
I have 8 chained 32x32 LED panels and I'm trying to display moving text separately in three areas of the frame:

        options.chain_length = 2
        self.left_matrix = RGBMatrix(options=options)
        options.chain_length = 4
        self.center_matrix = RGBMatrix(options=options)
        options.chain_length = 2
        self.right_matrix = RGBMatrix(options=options)
        
        left_matrix = self.left_matrix.CreateFrameCanvas()
        center_matrix = self.center_matrix.CreateFrameCanvas()
        right_matrix = self.right_matrix.CreateFrameCanvas()
        data = ["197", "text to display in a center", 3"]
        self.run_x_pos = center_matrix.width
        while True:
            self.left_matrix.Clear()
            self.center_matrix.Clear()
            self.right_matrix.Clear()
            self.textColor = graphics.Color(randrange(255), randrange(255), randrange(255))
            graphics.DrawText(left_matrix, font, self.run_x_pos, 22, self.textColor, data[0])
            text_len = graphics.DrawText(center_matrix, font, self.run_x_pos, y_pos, self.textColor, data[1])
            self.run_x_pos -= 1
            if self.run_x_pos + text_len < 0:
                self.run_x_pos = center_matrix.width
            graphics.DrawText(right_matrix, font, self.run_x_pos, y_pos, self.textColor, data[2])
            self.left_matrix.SwapOnVSync(left_matrix)
            self.center_matrix.SwapOnVSync(center_matrix)
            self.right_matrix.SwapOnVSync(right_matrix)
            
It doesn't work as it should, how to create multiple matrix correctly?

I found that it is possible to create multiple panels in https://github.com/hzeller/rpi-rgb-led-matrix/issues/622#issuecomment-389354771 comment

raspberry-pi-maker, 2021-07-19:
I think you can write multiple text lines in one canvas.
So there is no need to divide it into multiple canvases.

```python
options.chain_length = 8
self.matrix = RGBMatrix(options=options)
canvas = self.matrix.CreateFrameCanvas()
data = ["197", "text to display in a center", 3"]
self.run_x_pos = center_matrix.width
    while True:
        self.matrix.Clear()
        self.textColor = graphics.Color(randrange(255), randrange(255), randrange(255))
        graphics.DrawText(canvas, font, self.run_x_pos, 22, self.textColor, data[0])
        text_len = graphics.DrawText(canvas, font, self.run_x_pos, y_pos, self.textColor, data[1])
        self.run_x_pos -= 1
        if self.run_x_pos + text_len < 0:
            self.run_x_pos = canvas.width
        graphics.DrawText(canvas, font, self.run_x_pos, y_pos, self.textColor, data[2])
        canvas = self.matrix.SwapOnVSync(canvas)

```

xXZyzzXx, 2021-07-19:
> I think you can write multiple text lines in one canvas.
> So there is no need to divide it into multiple canvases.
>
> ```python
> options.chain_length = 8
> self.matrix = RGBMatrix(options=options)
> canvas [...]

I need the text to scroll only within its border. With one matrix, the pixels go beyond these limits and disappear only with a minus value of x.

raspberry-pi-maker, 2021-07-19:
I am familiar with image processing libraries such as PIL and OpenCV.
So, if I were to work, I would create 3 PIL images (64X32, 128X32, 64X32), work with text on each image, and then merge the 3 images.
And we will display the combined image on one canvas like this.

```python
from PIL import Image, ImageDraw, ImageFont

def get_concat_h(im1, im2, im3):
    dst = Image.new('RGB', (im1.width + im2.width + im3.width, im1.height))
    dst.paste(im1, (0, 0))
    dst.paste(im2, (im1.width, 0))
    dst.paste(im3, (im1.width + im2.width, 0))
    return dst

while True:
  im1 = Image.new('RGB', (64, 32),(255, 255, 255))
  im2 = Image.new('RGB', (128, 32),(255, 255, 255))
  im3 = Image.new('RGB', (64, 32),(255, 255, 255))
  fnt = ImageFont.truetype('./font/yourfontname.ttf', 40)
  draw1 = ImageDraw.Draw(im1)
  draw1.text((30,30), u'Hello World', font=fnt, fill=(0,0,255,128))
  draw2 = ImageDraw.Draw(im2)
  draw2.text((30,30), u'Hello World', font=fnt, fill=(0,0,255,128))
  draw3 = ImageDraw.Draw(im2)
  draw3.text((30,30), u'Hello World', font=fnt, fill=(0,0,255,128))

  im_pil = get_concat_h(im1,im2,im3)
  canvas.SetImage(im_pil)
  canvas = matrix.SwapOnVSync(canvas) 
```

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin