hzeller/rpi-rgb-led-matrix #1318: Problem with Pixel Mapping

Fipschen, 2021-07-03:
Hey together,
yesterday arrived my new Panels from China: https://de.aliexpress.com/item/32808141440.html?spm=a2g0s.12269583.0.0.46862570xkoCNR
I'm also using this Matrix Controller: https://joy-it.net/de/products/RB-MatrixCtrl

With my old Matrix it works fine to show the correct output, but with the new I have problems with the Pixel Mapping.

I've tried all combinations with --led-multiplexing und --led-row-addr-type, but I hadn't success. The best results did I get with multiplexing=1 and row-addr-type=0.
Could someone help me to find the correct Maaping?

Thank you very much.

Fipschen, 2021-07-03:
https://user-images.githubusercontent.com/83842738/124347372-b0ca3380-dbe4-11eb-9c1f-337784367c80.mp4

xXZyzzXx, 2021-07-03:
These setting helped me: -led-slowgown-gpio = 2 and -led-multiplexing = 2 for normal displaying

Fipschen, 2021-07-04:
> These setting helped me: -led-slowgown-gpio = 2 and -led-multiplexing = 2 for normal displaying

Hey,
thank you.
Slowdown has to be set as 4, with flickering I don't have problems. The Mapping of the Pixels is the Issue.

Fipschen, 2021-07-04:
I've found the solution in the documentation (why didn't I found it yesterday..?)
I need this parameters:
`--led-rows=32 --led-cols=32 --led-chain=2 --led-multiplexing=1`

closed 2021-07-04 by Fipschen