hzeller/rpi-rgb-led-matrix #1321: I made 384X512 pixel size LED display using Hzeller's library

raspberry-pi-maker, 2021-07-19:
I used 3X 8 P2.5 (64X128) pannels to make 384X512 size LED display.
My blog https://iot-for-maker.blogspot.com/ explains the detailed creation method and process. And the source code is also uploaded to github.

[image]

gregtap, 2021-07-27:
Awesome work! You should add some of your animations in Python samples too.
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python/samples

raspberry-pi-maker, 2021-07-27:
Hi Gregory.
In my github, test_video2.py shows video streamming.
I also added sample video file(souffle.mp4)

marcmerlin, 2021-08-02:
@raspberry-pi-maker  I didn't find the page that explains the wiring and how many rPis are used. I thought you could not go much past 384x320 with 3 channels and a single rPi
http://marc.merlins.org/perso/arduino/post_2020-03-13_RGB-Panels_-from-192x80_-to-384x192_-to-384x256-and-maybe-not-much-beyond.html

Also, issues are not the right place to discuss this, can you post on https://rpi-rgb-led-matrix.discourse.group/ ?
I would love to see what layout you have and how many channels.

marcmerlin, 2021-08-02:
Also after you move to https://rpi-rgb-led-matrix.discourse.group/ , please put a link to your post here and close the issue :)

raspberry-pi-maker, 2021-08-03:
Hi @marcmerlin
I' close this issue and move to https://rpi-rgb-led-matrix.discourse.group/t/i-made-384x512-pixel-size-led-display-using-hzellers-library/292

All of my blog posts related to this LED sign are made up of six. In 6 articles, the manufacturing process and necessary parts were explained.
I used a total of 5 Raspberry Pis. One was used as an image or video streaming server and the other four were used for LED displays. So each player controls 6 LED panels. Streaming server and player send and receive data via UDP. For more details, please refer to the 6th blog post and the code on Github.

closed 2021-08-03 by raspberry-pi-maker