hzeller/rpi-rgb-led-matrix #1322: Text glitching at half of the panel

jesusmsr, 2021-07-21:
So I connected a 64x32 matrix to my raspberry pi 3 b+, ran the sample files but it is acting weird.
I got 2 panels and I tested both and it does the same with both, so I assume is not a hardware problem.
I'm also using Adafruit's bonnet.
Here is a video of what the issue looks like

https://imgur.com/a/1ROZ5p0

Beersatron, 2021-07-21:
What does the back of the panel look like? Can you try it without the bonnet?

The ElectroDragon bonnet might be a better alternatively, if you can get one https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

jesusmsr, 2021-07-21:
This is how it looks https://imgur.com/a/Fqy16TH
I tried connecting directly gpios to the connector in the panel with jumper cables but it was a mess and I didn't manage to do it.

jesusmsr, 2021-07-22:
I have to add: running the pulsing colors sample works, it displays colors correctly, it is just with the running text and square samples that it does this kind of glitching

giuseppedevito, 2021-08-12:
Can you write the RGBMatrix options that you are using?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin